prusa3d/PrusaSlicer #6828: Line-Width can't go above 1mm

LordRofl, 2021-08-23:
### Version
_Version of PrusaSlicer used goes here_
2.3.3

### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
ELEGOO Neptune 2 (Marlin)

### Behavior
If i go above 1mm width in Printing Settings - Advance lets say 1.12 its capped on 1mm when i change to the preview. 

I was able to setup 0.2, 0.4, 0.6 and 0.8 just fine. My Linewidth is 1,125 x the nozzle diameter
and i got for example with 0.8 a correct width under the slicer preview about 0.9 mm.

When i try to setup this exact thing with 1.0 nozzle it is stuck at 1mm under the slicer preview.

LordRofl, 2021-08-23:
Okay got it, since it ignores my minimum height setting it needs to be increased in layer height.

LordRofl, 2021-08-23:
After thinking about it a bit, I do have to open this up again. The ratio between layer height and line width should not be so important as the diameter of the nozzle increases. I'm sure I can stay in vase mode at 0.2mm layer thickness no matter what nozzle size. This should not be denied to me by limitations in the software.

In my case, I can't run a line width of 1.12mm with a layer thickness of 0.2mm in vase mode with a 1mm nozzle. This would certainly be possible without problems.

bubnikv, 2021-09-25:
Toolpath width in G-code Preview was clamped to 1 mm, which did not work for large nozzle diameters. The width is newly clamped to 2 mm or 4 * layer height, whatever is bigger.

closed 2021-09-25 by bubnikv