prusa3d/PrusaSlicer #6832: Display count of tool changes without wipe tower enabled

ksipple, 2021-08-24:
### Version
2.3.3+

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S+

### Behavior
Number of tool changes is only shown in statistics when a wipe tower is enabled. This should not be based on if the wipe tower is enabled, it should check for extruders being greater than 1 instead. When doing manual filament swaps I don't always need a wipe tower but would like to see the number of tool changes needed.

lukasmatena, 2021-08-25:
You are right. The same applies to `total_toolchanges` placeholder.

mrhasbean, 2022-04-10:
Yes, both should also default to 0 for single extruder prints.

themacboy, 2022-07-25:
Good morning,
I just finished my IDEX mod, but I can't find a way to specify in the gcode if the slice only has one or two toolheads. 

I tried using "total_toolchanges", but as explained before, it only works with a wipe tower, and personally for my Idex printer I don't need to use a wipe tower.

Is there a way to know if one or two heads will be used in the slice?

thanks in advance!

(sorry for my poor english)

themacboy, 2022-07-25:
I answer my self ... with one possible option ... 

set wipe tower width and brim to 0 

;)

ddm-j, 2022-09-06:
Chiming in. I've written a custom GCODE command to prime both extruders at the start of a IDEX print. Unfortunately this code hinges on `total_toolchanges` working properly without the wipe tower enabled. At this time it does not. 

1. `total_toolchanges` should not depend on the wipe tower function
2. "Prime Extruders" should be made available to all IDEX prints regardless of wipe tower. Gets rid of my need for custom start gcode.

andreas-rapid-ood, 2022-10-25:
I have the same issue. My tool change script needs the info if a tool changes takes place or not. All printers which have several tools but do not use prime towers (but purge buckets for example) should have this issue if they want to use one profile for single material and multi material prints

apphead, 2023-01-24:
Shame this does not get fixed. 
This line of code is just not right:
https://github.com/prusa3d/PrusaSlicer/blob/43f2018062a4c29fac7e4f89348dc5c0536657e2/src/libslic3r/GCode.cpp#L950

RobOvalle, 2023-10-05:
> I answer my self ... with one possible option ...
> 
> set wipe tower width and brim to 0
> 
> ;)

This crash the software.

lukasmatena, 2023-10-06:
We will look into it soon.
SPE-1956

lukasmatena, 2023-11-11:
Fixed in [2.7.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-beta1). Closing.

closed 2023-11-11 by lukasmatena

Lyxar13, 2023-12-25:
It is not fixed (V1.8.2.56)! :-( I tried it today and it looks like the same issue as written in this post. It depends on the wipetower if total_toolchanges works or not. 
[image]
[image]

Start GCODE for the purge Lines:

{if total_toolchanges==0}
T{initial_extruder}
G92 E0
G1 F1200 X5 Z0.5
G1 F600 X100 E20
G1 F600 Z2
G92 E0
G1 F1200 E-6
{if initial_extruder==0}G0 X-20 F12000 ; park T0 {endif}
{if initial_extruder==1}G0 X350 F12000 ; park T1 (old 345) ; park T0 {endif}
{endif}
{if total_toolchanges>0}
T0
G92 E0
G1 F1200 X5 Z0.5
G1 F600 X100 E20
G1 F600 Z2
G92 E0
G1 F1200 E-6
G0 X-20 F12000 ; park T0
T1
G92 E0
G1 F1200 X295 Z0.5
G1 F600 X200 E20
G1 F600 Z2
G92 E0
G1 F1200 E-6
G0 X350 F12000 ; park T1 (old 345)
{endif}

lukasmatena, 2024-09-12:
Broken again in 2.8.0, will be fixed in 2.8.1. Sorry.

lukasmatena, 2024-09-19:
Fixed again in 2.8.1-rc1.