prusa3d/PrusaSlicer #6834: Gaps in object are not filled in 2.4a

davidkerny, 2021-08-25:
The same project is sliced differently in 2.4a unlike the earlier Slicer versions.
The model has two parts next to each other. PS2.3.3 fills the gap between two parts but I can't find a way to fill the gap using the 2.4. Slicing mode doesn't change it. IS there a way to make the 2.4 behave the same as the earlier versions did?

[image]
[rozdeleni_test.zip](https://github.com/prusa3d/PrusaSlicer/files/7045888/rozdeleni_test.zip)

TojikCZ, 2022-01-11:
What speed do you have for gap fill in Print settings -> speed?

davidkerny, 2022-01-17:
Gap fill is 30mm/s - doesn't seem to affect the slicing preview too much.
The outcome can be fixed by altering the model - if there is an overlap between the two parts of the object, it slices as it used to in 2.3.3 - and the two parts act as a single object in terms of slicing.

If there are two parts, one next to the other with no overlap - the Slicer makes the external perimeters in between these.
The division of the object into the parts was made in Fusion360 and theoretically, there shouldn't be a gap in the model - although the F360 is able to generate weird STLs on occasion. The behavior changes slightly when altering the Slice resolution but doesn't seem to fill the potential void entirely using any reasonable setting.

closed 2026-09-29 by prusaslicer-bot[bot]