prusa3d/PrusaSlicer #6835: Feature Request - Color Picker Hex Codes

ksipple, 2021-08-25:
### Version
2.3.3+

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
MK3S+

### Behavior
In the filament color picker please add the ability to enter and view color hex codes. The slicer uses these in the background for example using the filament_colour placeholder in custom g-code. It would be nice to be able to specify the hex code without converting values with a 3rd party tool every time. I convert the hex codes to color names in the custom g-code to display on the printer when switching filaments.

andrhamm, 2024-03-28:
I just encountered a need for this too, the basic color picker, at least on macOS, is frustratingly limiting!

saw2kx, 2025-03-24:
It seems like Prusa want this too as they only provide hex codes for colours on their filament information pages.

closed 2026-09-29 by prusaslicer-bot[bot]