prusa3d/PrusaSlicer #6836: Ender5 x and y axes are inverted

nordurljosahvida, 2021-08-25:
### Version
2.3.1

### Operating system type + version

Ubuntu 20 GTK-3 appimage

### 3D printer brand / version + firmware version (if known)

Creality Ender-5

### Behavior
* _Describe the problem_

Ender 5 has home in the back-right. Print coordinates X10Y10 are in the back right, so print comes out rotated [not mirrored, the object is correct, just the orientation is wrong in the virtual buildplate].

* _Steps needed to reproduce the problem_

Use bundled Ender 5 profile

_Is this a new feature request?_

No

I know we can tell PS home position offset, but we cannott affect orientation, right?

rtyr, 2021-08-25:
It seems like misconfigured firmware to me, is it stock firmware? If it physically homes back-right corner by design, it doesn't automatically mean it is 0,0 position (and I don't see any reason why would anyone want 0,0 in back-right corner). For example, Prusa MINI homes X axis to the right side, but 0 position is still on the left side. 

Anyway, Creality Slicer with Creality profile considers left-front corner as 0,0 for Ender 5, so I think we shouldn't change anything.

FYI @pmjdebruijn

khseal, 2022-11-30:
I have the same problem with another printer. It's just convenient for me to have parking on the other side. Other slicers have axis inversion =(