prusa3d/PrusaSlicer #6847: Feature Request

MrCroup, 2021-08-28:
### Version
Latest

Add the ability for custom gcode to recognize PRINT_START/START_PRINT macro or, as in SuperSlicer (based on PrusaSlicer) add a function to only use the custom start gcode with no modifications by PrusaSlicer.

I would like to use PrusaSlicer for my 3D Printers that are using Klipper firmware, but PrusaSlicer doesnt recognize the PRINT_START/START_PRINT macro as a valid startup gcode and adds gcode to set the temperature for extruder and bed.

With klipper, the macro is used to bring the heat up in stages and allow for heat soak time without bringing the system up to 100% temp on the extruder. A typical Klipper START_PRINT macro would look like this in the start custom gcode:

START_PRINT T_BED=[first_layer_bed_temperature] T_EXTRUDER=[first_layer_temperature]

rtyr, 2021-08-29:
I believe this is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2420. This workaround should work for you https://github.com/prusa3d/PrusaSlicer/issues/2420#issuecomment-608255455.

lukasmatena, 2023-04-02:
Setting temperatures automatically is optional since [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena