prusa3d/PrusaSlicer #6851: Opening a 3MF from linux file explorer doesn't load project name

nordurljosahvida, 2021-08-29:
### Version
2.3.1

### Operating system type + version
Ubuntu 20

### Behavior

Opening a 3MF file from ubuntu 20 file explorer [prusaslicer made default systemwide for 3MF]. 
Save-as.
Name suggested is not the filename as should be. Instead the name of one of the geometries inside is suggested.
Opening a 3MF via ctrl-o from an already running PS instance isn't affected by this, and in fact correctly suggests the filename as the name for the file to be saved.

_Is this a new feature request?_
sort of

tamasmeszaros, 2021-09-01:
Hi @nordurljosahvida ! I can confirm this issue on Manjaro with PS v2.3.3. It seems that the problem is not present anymore with latest master.

bubnikv, 2021-09-01:
@nordurljosahvida  How did you install PrusaSlicer? Where did you get it from?

nordurljosahvida, 2021-09-02:
@tamasmeszaros good to know thanks. @bubnikv sorry forgot to mention, GTK3 appimage from github releases page.

bubnikv, 2021-09-03:
@nordurljosahvida Please run the latest 2.4.0-alpha1 and try the integrated desktop integration.

MNeMoNiCuZ, 2021-09-04:
I have a similar issue on Windows 7.
It may be related or not. I can add it as a separate issue.

**3mf file format opening does not open it as a "project"**

- If you have assigned the .3mf-format to the PrusaSlicer software, and you double-click on a file, it will open the .3mf, but it won't have said 3mf open as a "project".
- This means the first time you go to File > Save, it will do a "Save As", even if you already have a project file opened.
- The name of the 3mf is not listed in the window name, it just says: "PrusaSlicer".
- If I do a "File > Open Project" and load a .3mf-file, it opens properly, and the window title is set to the name of the 3mf-file.

bubnikv, 2021-09-06:
@MNeMoNiCuZ Would you please check with PrusaSlicer 2.4.0-alpha1? Thanks.

MNeMoNiCuZ, 2021-09-06:
Seems to work well @bubnikv.
Tested on the PrusaSlicer-2.4.0-alpha1+win64-202109011441.zip build on Windows 7.

nordurljosahvida, 2021-09-06:
> @nordurljosahvida Please run the latest 2.4.0-alpha1 and try the integrated desktop integration.

works, thanks!

nordurljosahvida, 2021-09-06:
@MNeMoNiCuZ closing as solved, thanks to all for work and testing

closed 2021-09-06 by nordurljosahvida