prusa3d/PrusaSlicer #6853: Vertical Shell Thickness not honoring checkbox

GWdd, 2021-08-30:
### Version
2.3.3

### Operating system type + version
Win 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa Mk3  v3.10 

### Behavior
Setting Ensure Vertical Shell Thickness to OFF (unchecked) does not affect slice. On steep slopes, the slicer is still placing copious infill when none is wanted.  In the online help, there is a blurb that says there is a known issue with this control, but it seems like a different issue than what I am dealing with here.

[image]
[image]

#### Project File (.3MF) where problem occurs
[EnsureVerticalShell-defect.zip](https://github.com/prusa3d/PrusaSlicer/files/7073560/EnsureVerticalShell-defect.zip)

rtyr, 2021-08-30:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/1054, https://github.com/prusa3d/PrusaSlicer/issues/3744, https://github.com/prusa3d/PrusaSlicer/issues/223, etc.

The "Ensure vertical wall thickness" feature was implemented as an alternate implementation of the solid infill on sloping surfaces. When the checkbox is unchecked, the old algorithm from upstream Slic3r is used.

We plan to look at it soon, we have some ideas on how to improve it.

Goury, 2022-01-06:
> We plan to look at it soon, we have some ideas on how to improve it.

You don't need to improve, this is an anti-feature.
You need to remove it.

Vertical wall thickness should only be ensured by adding extra perimeters, not infill.

simonkuehling, 2022-05-20:
To add another good reason to be able to acutally disable the feature: It can really mess with anchoring of bridging solid infill on top of sparse infill: The bridging infill does not anchor to the perimeter if there is this excess solid infill being build around the edges:

Order of operations in three steps:
[image]
[image]
[image]

Print result - which failed due to collision with curled up material blobs from the bridging infill:
[image]

knightsljx, 2023-02-02:
This issue still has not been fixed in 2.5.0, Jan 2023. Unchecking the box still generates the purple solid infill. This should be a high priority fix as it severely impacts print time and material cost. People who are unaware of the height range modifier workaround from #8727 are basically wasting hours of their lives and tons of material printing unnecessary infill. Thank you.

Goury, 2023-02-02:
@knightsljx it probably won't be ever fixed, because otherwise Prusa will have to admit that his slicer was broken for years and all the influencers who praise this broken thing will have to admit their utter incompetence and being sold out.    
Their combined ego can not take it.    
Not only that, but Prusa doesn't care about doing the right thing, he cares about underpaying his employees while creating a cult around his name.

kebes22, 2023-02-17:
So, I just switched from Simplify3D, partly because it also has this problem, and also I have been looking for a good free slicer with regular updates (S3D barely meets one of those criteria). I have been frustrated to find that this is also a problem here. My issue, however, isn't about speed or shaking or filament use as others are complaining about. My problem is that I am trying to print extremely soft TPU parts with only single outer walls, and this setting is adding a crap ton of material to my sloping walls. the best I can get is about equivalent to ~3 walls thick, which makes those parts of the print EXTREMELY stiff (compared to the 1-wall sections). Also it makes NO sense that enabling "ensure vertical shell thickness" would actually be the best option for this. Please fix this soon so I can finally get the quality of soft TPU prints I have been hoping for.

dehmlowm, 2023-02-28:
I think that issue 223 is the original issue. I pretty sure that this problem didn't exist in PrusaSlicer at some point and 2017 seems like the right time frame that is showed up. I will make the same offer here that I will Venmo someone who fixes this bug $500 US:
https://github.com/prusa3d/PrusaSlicer/issues/9245 -includes some test file that easily reproduces the issue and reproduction steps.

Probably original issue
https://github.com/prusa3d/PrusaSlicer/issues/223

Goury, 2023-02-28:
@dehmlowm not really.
Back then thickness ensurer was just a useless anti-feature.
The actual real problem is that now (since some more recent time) you can't turn it off unless you also turn off infill.

To add to your collection, it doesn't actually ensure wall thickness if you have thin walls in your model.
One would expect that walls thinner than extrusion diameter multiplied by wall count would be enthickened to that minimum value, but such walls stay thin.

Here's an example:
[image]
[image]
[image]
There is a wall in there, it's just very thin.

dehmlowm, 2023-02-28:
> you can't turn it off unless you also turn off infill.
> To add to your collection, it doesn't actually ensure wall thickness if you have thin walls in your model. One would expect that walls thinner [...]

@Goury 
Thanks for adding to the pile!  I did include some example files in 9245 that have infill even if infill is set to 0%

@rcmz confirmed that this is still in 2.6 unfortunately

knightsljx, 2023-04-02:
years later, this serious problem is still unfixed. I refuse to buy any more Prusa products until this problem is resolved

lukasmatena, 2024-11-21:
"Ensure vertical shell thickness" option is restored in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) and hopefully configurable enough to cover all use cases.  Closing.

closed 2024-11-21 by lukasmatena