prusa3d/PrusaSlicer #6854: Feature Request: Print *all* perimeters before infill

TickTock-, 2021-08-30:
### Version
2.3.3

### Operating system type + version
Win10 64

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
I often edit the gcode to perform a color change after perimeters are complete.  This is typically to enhance the visibility of graphics, text, etc.  Pretty straight-forward when only one object is in the plane.  However, when there are multiple objects, the slicer prints each in turn.  This requires either 1) multiple color changes for each object (8 color changes in the screen shot below where you can see it has started infill before printing perimeters on the other objects) or 2) manually editing the gcode to put all the perimeters together.  The latter is doable but a little tricky because of the wipe and travels between objects.  I would like an option to have the slicer print all perimeters first before starting infill to make the m600 instertion easier.

This request becomes more important once [Feature Request #5195](https://github.com/prusa3d/PrusaSlicer/issues/5195) is implemented (within layer color change).

[image]

rtyr, 2021-08-30:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2967.

rtyr, 2021-09-01:
No response, closing as a duplicate.

closed 2021-09-01 by rtyr