prusa3d/PrusaSlicer #6855: PrusaSlicer will not run on my Windows laptop

pigskins65, 2021-08-30:
### Version
Latest 2.3.3 version from website and Git. Also tried 2.3.2

### Operating system type + version
Windows 10 64-bit (latest updates)

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
Installs successfully but double-clicking icon to open it does nothing. Mouse pointer spins for a sec or 2 then stops. I have tried to see what is happening in task manager and one time I did see "PrusaSlicer suspended".

pigskins65, 2021-08-30:
Update: I discovered that this occurs if I connect a second external monitor via USB3 to HDMI and disable the laptop screen. If I have the laptop screen on and duplicate it on one of the external monitors it will run. Strange?

bubnikv, 2021-08-31:
Your issue is similar to one of those:
#6412
#5573
#6412 
#5573
#6710
#5518
#4871
#3726

We encountered two devices in house (600+ workers): One was a windows laptop with a docking station where Slicer no more crashes after some update. The other is an USB docking station where Slicer behaves weirdly only with some configuration of monitors (two particular monitors, both the same). We already debugged that configuration and it seems PrusaSlicer crashes inside an OpenGL driver, thus we don't know much more than before. We will invest more time into investigating the issue on that particular setup though, we hope to fix it and that fixing that particular setup will solve many of the issues above.

pigskins65, 2021-08-31:
Interesting! Well if there's anything I can help with in regards to providing the product info or driver info I'm happy to do so. I appreciate the update.

lukasmatena, 2023-02-22:
@pigskins65 is this still happening with 2.6.0-alpha4?

closed 2026-09-29 by prusaslicer-bot[bot]