prusa3d/PrusaSlicer #6857: layer_num does not work over 3734

SirColin, 2021-08-30:
### Version
2.3.3+x64

### Operating system type + version
macOS big Sur 11.5.2

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3s+ 3.10.0-4990

### Behavior
When adapting Gcode before changing the layer with the parameter {if layer_num == 3734} X000 X000 {endif} it works, but as soon as I go over layer 3734 this is no longer output in gcode.

I have to say that I use the sequential printing function with a spiral vase.

I have tried several other STL files, but from layer_num 3735 onwards it is over every time.

* Steps needed to reproduce the problem

1. activate complete individual objects
2. activate spiral Vase mode
3. add before layer change gcode `{if layer_num == 3734} M106 S255 ;you will find me in the gcode{endif} {if layer_num == 3735} M106 S255 ;you will find me NOT in the gcode{endif} `
4. generate the gcode an inspect with editor

foreachthing, 2021-08-31:
Try {if layer_num > 3734}.

lukasmatena, 2024-02-19:
Cannot reproduce, likely outdated. Closing.

closed 2024-02-19 by lukasmatena