prusa3d/PrusaSlicer #6858: Unable to compile vertex shader of shader program 'gouraud_light_instanced'

wavexx, 2021-08-31:
### Version
Current master: version_2.4.0-alpha0-1298-g0e428eb41

### Operating system type + version
Linux: x1/debian unstable

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
During startup PrusaSlicer shows a dialog about "gouraud_light_instanced not found". It does exist, but fails to compile:

```
Unable to compile vertex shader of shader program 'gouraud_light_instanced':
0:18(1): error: `in' qualifier in declaration of `v_position' only valid for function parameters in GLSL 1.10
0:19(1): error: `in' qualifier in declaration of `v_normal' only valid for function parameters in GLSL 1.10
0:21(1): error: `in' qualifier in declaration of `i_offset' only valid for function parameters in GLSL 1.10
0:22(1): error: `in' qualifier in declaration of `i_scales' only valid for function parameters in GLSL 1.10
```

_Is this a new feature request?_

No

#### Project File (.3MF) where problem occurs
N/A

enricoturri1966, 2021-08-31:
This should have been already fixed by https://github.com/prusa3d/PrusaSlicer/commit/c797bed67d0db1281fb0820238aa811a4b66fcdf

wavexx, 2021-08-31:
:zap: indeed! Faster than I could update ;)

closed 2021-08-31 by wavexx