prusa3d/PrusaSlicer #6860: Status bar text colour in dark mode

jokohoko, 2021-09-01:
### Version: 2.4.0 Alpha

### Operating system type + version: Windows 10

### Behavior:

The status bar in the bottom of the screen is hardly readable in dark mode. I love how it looks, but this is the only mistake i can find in the color scheme.
[image]

gudnimg, 2021-09-01:
Minor issue related to this, I think these arrows should be white:

[image]

[image]

jokohoko, 2021-09-01:
And as i use it further also these buttons have little contrast. 
[image]
Compared to the plater and print settings on the left that seem like the same type of buttons i prefer the orange-ish background over the bold font to indicate selection
[image]

bubnikv, 2021-09-02:
Thanks for your feedback.

@YuSanka spent considerable time trying to dark-mode some of the Win32
controls. For the menu arrow there is likely no way to do that, the same
applies to the status bar.

If we are lucky, we will get rid of the status bar and replace it with a
progress bar notification. I am afraid we will not be able to fix the menu
arrow as it is being painted by Windows.

YuSanka, 2021-09-08:
> And as i use it further also these buttons have little contrast.
> [image]
> Compared to the plater and print settings on the left that seem like the same type of buttons i prefer the orange-ish background over the bold font to indicate selection
> [image]

Ok, @jokohoko, which variant do you prefer? :) 

[image]
or 
[image]

IMHO, first variant is more observed but it looks like next tab in Notebook and can be confusing for some one.

jokohoko, 2021-09-10:
Sorry, github notifications came 2 days late.
I'm not sure if I am missing the reason for this. But why do the buttons on the right have a different style as the ones on the left?
The first one has my preference since the orange border is the best indication to me. But on the first i would drop the boldness on the selected one. As far as I can find no other element has boldness to indicate something.
Also on the normal light mode i would drop the boldness on the right buttons.

bubnikv, 2021-09-25:
The left "buttons" are notepad tabs. The right "buttons" are radio buttons, thus they are different controls.

The status bar was removed, the hightlight at the mode selector buttons was improved. 

I am afraid we will not be able to fix the menu arrows }thanks Microsoft not), but I don't think this is a big deal.

[image]

closed 2021-09-25 by bubnikv