prusa3d/PrusaSlicer #6863: "Support on build plate only" no longer overridden by support painting

MarkWheadon, 2021-09-01:
### Version
_Version of PrusaSlicer used goes here_
2.4.0-alpha1
### Operating system type + version

MacOS

### 3D printer brand / version + firmware version (if known)
### Behavior

In previous versions of PrusaSlicer, if you had "support on build plate only" set, you could still override this behaviour by painting the support on. So the blue on the sphere in this:

this:
[image]

got you this:
[image]

but now I get this:
[image]

(i.e. that blue painting in the support painting mode is ignored.) Is this change in behaviour intentional? (I hope not – the old behaviour is really useful.)

Mark

rtyr, 2021-09-01:
Please provide your 3MF project file. It is multipart object, right?

MarkWheadon, 2021-09-01:
I may have to fabricate one as this is a custom model. It's not properly
multi-part. It's little discs merged with the main object.

Mark

MarkWheadon, 2021-09-02:
Ah! On setting this up with a simple sphere and cylinder I now know why other people haven't seen this. I was being dumb, but hey, it's exposed a bug.

So, being sensible with the support painting (still with support only on bed ticked), this:
[image]
gets this, which is what you'd expect:
[image]
But I painted the top of the cylinder as well (shouldn't make a difference?):
[image]
which gets you this (which I would think is wrong?):
[image]
And I'll attach the demo 3MF file.
[Demo.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7098681/Demo.3mf.zip)
(zipped to allow the upload)

lukasmatena, 2021-09-02:
@MarkWheadon The Demo.3mf file (the second example) is sliced correctly. You have 'Supports on build plate only', so they are on build plate only. Removing the painting changed nothing when I tried.

Can you please enclose the 3MF of the first example that you mention, one that you think is broken? Does it also have 'Supports on build plate only' selected? Thanks.

lukasmatena, 2021-09-02:
@MarkWheadon (I deleted my previous post which was wrong).
I don't see the dependence on whether the top of the plate is painted or not, but this really is a bug, support enforcers should take precedence. We'll look into it, thanks.

MarkWheadon, 2021-09-02:
Good stuff – you have to remove every last crumb of the blue on the cylinder for it to start behaving including that which is underneath the sphere – so click on the cylinder in the right-hand panel (so you've selected just the cylinder and not the whole merged object), move the cylinder in the left-hand view to expose it completely, click on the object in the left-hand view to select the whole merged object, unpaint all the blue on the now exposed cylinder top with shift-right, then click on the cylinder in the right-hand view, drag it back in the left-hand-view to where it was, and then it'll have support everywhere.

Mark

closed 2021-09-02 by bubnikv