prusa3d/PrusaSlicer #6864: Crash when opening on Second Monitor, Asus sonic suite 2 cause, no "Nahimic" service or warning

werickson24, 2021-09-01:
### Version
PrusaSlicer 2.4.0 alpha 1

### Operating system type + version
Windows 10  10.0.19042 Build 19042.1165 20H2

### 3D printer brand / version + firmware version (if known)
MK3s+ & MK2.5s MMU2s

### Behavior
When Opening PrusaSlicer on my second monitor by either default or dragging the open window to the second monitor and closing it, then reopening it OR doing anything in it after dragging it to the second monitor, it freezes up and then must be force closed.

This doesn't look like Nahimic, as it does not appear in Task manager, AND **I do not receive any warnings about an unsupported DLL**
[image]
[image]

**HOWEVER, I do have an ASUS board with their sonic suite 2 running, and when closing it from the system tray PrusaSlicer works!**
This appears to be another service, Maybe Nahimic but renamed as when I dug through the sonic suite 2 program folder and sure enough, 
[image]
and 
[image]

Since it seems this issue is not fixable by the Awesome PrusaSlicer dev's and that we'll have to rely on sketchy other dev's, i'm mainly making this issue so we can add whichever DLL this is to the blacklist.

werickson24, 2021-09-03:
Here's the only two DLL files that are not just a Windows system DLL or a PrusaSlicer DLL
[image]

bubnikv, 2021-09-03:
Thanks. SS2OSD.dll sounds similar to the Nahimic OSD. OSD means On Screen
Display I suppose, that is likely the module that collides with our
application. We will add it to our black list.

kocikdav, 2021-09-03:
Hello @werickson24 please try this [build](https://www.dropbox.com/s/i7xr29bhunvnlh1/PrusaSlicer-2.4.0-alpha1%2B23-win64-ge630332e2-202109030730.zip?dl=0). It should show you warning about SS2OSD.dll being injected into PrusaSlicer.

werickson24, 2021-09-03:
Works perfectly! 
[image]

bubnikv, 2021-09-06:
Great, thanks for testing. Your check will be in the next alpha.

closed 2021-09-06 by bubnikv