prusa3d/PrusaSlicer #6865: Release version 2.4 as version 3.0

nsheridan, 2021-09-01:
### Version
2.4-alpha1

### Behavior
The amount of fixes and features gone into the 2.4 alpha is so extensive that it should warrant a 3.0 release.

There's a risk that a future major number bump to 3.0 would make no sense, as the amount of features and fixes wouldn't warrant it and Prusa Slicer would be stuck on 2.x forever 😛

ssill2, 2021-09-01:
I would second this.  The set of changes vis 2.3.x is extensive.

th33xitus, 2021-09-03:
I don't think this issue tracker is the right place to discuss how they call their versions.
And if i am not completely wrong, they are following the semantic versioning system as there are some references to it in the sourcecode. In that case calling it 2.4.0 is reasonable.

rtyr, 2021-09-03:
I take this issue as positive feedback. :-)
Don't worry, we have still plenty things to implement and to fix.

I think we can close this.

closed 2021-09-03 by rtyr