prusa3d/PrusaSlicer #6868: FEATURE REQUEST Enable Bottom contact Z distance in soluble supports

Maciey2, 2021-09-01:
### Version
2.4.0-alpha1

### Operating system type + version
Win64

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
FEATURE REQUEST

I use the soluble supports option, but instead of using soluble filament I use a marker on the last layer of supports
This makes the first layer on the supports come out very nice and the supports (of the same material ) are easy to remove.
When the supports have to start on the model I also have to paint a piece of the model with a marker where the supports will be.
It is very hard to judge where the model peremiters will start to be printed and where the supports will be, to paint them with a marker.

In the latest version of prusaslicer there is an option to select the distance of the bottom layer of supports from the model, "Bottom contact Z distance" but with the option of soluble  supports it is greyed out.

It would be very cool if this option was also available for soluble supports.

[image]
[image]

rtyr, 2021-09-15:
Sorry, I don't think your unusual use case is enough to justify breaking change like this. It is also not as simple as you may think, because support layers are synchronized with object layers in "soluble mode" (contact Z distance = 0). When contact Z distance is not 0, support layers are printed with variable layer height, independent on objects layer height.

That being said, I may have a simple workaround for you, which should work in 2.4.0-alpha1:

1) Disable `"Thick bridges"` parameter (so 1st layer printed on the supports will be printed with regular flow - similar to flow used with contact Z distance = 0 (soluble)). 
2) set your `"Bridge flow ratio"` to 1
3) set top Z contact distance to very low value, for example 0.01mm

This way, you should be able to set different bottom Z contact distance without quality loss on top side.

Maciey2, 2021-09-15:
Thank you for this advice. This is exactly what I was looking for. I did not know that you can set such a small gap as 0.01 mm

rtyr, 2021-09-15:
Don't forget to disable "thick bridges". I believe we can close this issue.

closed 2021-09-15 by rtyr