prusa3d/PrusaSlicer #6871: Strange Slicing after cuting a model in Prusa Slicer 2.3.3

alexnet1977, 2021-09-02:
### Version
2.3.3 Win 64
### Operating system type + version
Windows 10 Pro 21H1
### 3D printer brand / version + firmware version (if known)
Prusa M3S FW 3.10

### Behavior
I modelled a Cable Protection with a M40x1.5 Thread with Inventor.
I loaded  it to Prusa Slicer, to get the cables through the 3 Holes, I divided it
with Prusa Slicers cutting function in the middle Line through all holes.
I put the two parts on the bed and sliced it. one part is just fine, in the other one there are
strange artefacts in the thread. I tryed it 3 times on 3 different Computers.

[image]

When I slice the model as one part, there is no such artefacts, as shown in the screenshot. 

The 3mf Project File with as shown in the screenshot

[M40-Cableprotection.zip](https://github.com/prusa3d/PrusaSlicer/files/7096289/M40-Cableprotection.zip)

YuSanka, 2022-02-03:
Your objects have errors after cut

[image]

If you click on exclamation icon then "Fix through NetFabb" will be processed. Fixed objects are sliced now without issue. :)

[image]

@alexnet1977 Can we close this issue?

rtyr, 2022-03-10:
Answered, no response. Closing.

closed 2022-03-10 by rtyr