prusa3d/PrusaSlicer #6873: App will not start if the bed texture image is not found

sanchosk, 2021-09-02:
### Version
PrusaSlicer-2.3.3+linux-x64-202107161032_82dd34f525fa1b930725b2d120730fdd.AppImage

_Use `About->About PrusaSlicer` for release versions_
Version 2.3.3+linux-x64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
N/A

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 20.04.02 64bit

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
N/A

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Ender 3 v2

### Behavior
* _Describe the problem_
When starting the app, it exits with error message terminate called without an active exception

* _Steps needed to reproduce the problem_

1. select a printer with specific custom texture
2. exit the app
3. rename or delete the texture file
4. start the app again

  * _If this is a command-line slicing issue, include the options used_
  N/A
* _Expected Results_
App starts OK, does not show custom texture

* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
App does not start, reports terminate called without an active exception, even loglevel 5 does not bring more info.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
N/A

FidelCapo, 2021-09-02:
I cannot reproduce this issue on Linux in version 2.4.0-alpha1.
Closing.

sanchosk, 2021-09-02:
Is it possible to reopen? I downloaded the 2.4 alpha appImage and I can still reproduce it - just removing one folder from the path of the file and bam - the app will not even start. If I do it for the printer that is not selected as the current, the app will start, but when switching the other (wrongly configured) printer, it will crash again.

sanchosk, 2021-09-02:
@FidelCapo I can provide the ini file for my printer, if that helps...
The log itself is not very helpful as the only difference in "good" or "bad" setup is the line with termination.

FidelCapo, 2021-09-02:
Yes, please attach zipped your ini, thanks.
We tried it also with same Linux OS, even with same printer profile but we still cannot reproduce it..

sanchosk, 2021-09-02:
[ender_broken_ini.zip](https://github.com/prusa3d/PrusaSlicer/files/7098696/ender_broken_ini.zip)

The problem is the file path - if I change it to the correct one (without users), it works. If I keep the users there (old installation) it does not start.
Fun fact - on the version prior to 2.3, the slicer started, but did not show the image.

Further observation:
If I start with correct ini file and then switch the printer to one with wrong path, there IS an error message in the log and with Alpha version also on the screen:
Internal error: boost::filesystem::status: Permission denied: "/home/klaucor/.config/PrusaSlicer/printer/enderV1bed.svg"

sanchosk, 2021-09-02:
Heh, I found the difference!
If the file does not exist, but the folder path does, the app will work and show no image of the print surface.
If the path is invalid, it will crash.

lukasmatena, 2021-09-02:
It should be fixed by 18c8204. Closing. In case it is not ok in the next release (2.4.0-alpha2), let us know. Thanks.

lukasmatena, 2021-09-02:
@sanchosk To be on the safe side, could you please test the following build? We are not entirely sure if we reproduced exactly the same problem that you are reporting.

https://drive.google.com/file/d/1-ZZOYWZfYJ8CWWfnt9x77oAa0sqStSw8/view?usp=sharing

sanchosk, 2021-09-02:
Confirmed - with the version PrusaSlicer-2.4.0-alpha1+1-lm-issue6783-linux-x64-ga1ce514-202109021350_59ea3a404a68e53ecc36c4e3dc37eb41.AppImage the app starts OK in both cases:

- when starting with incorrect profile, app starts OK, printer loads, shows no graphics on bed
- when starting with correct profile and then switching to incorrect, the app keeps working and shows no graphics on bed

Thanks a lot for the fix!

lukasmatena, 2021-09-02:
@sanchosk Splendid. Thanks for testing. Closing (again).

closed 2021-09-02 by lukasmatena

Felix14-v2, 2024-07-02:
> It should be fixed by [18c8204](https://github.com/prusa3d/PrusaSlicer/commit/18c8204837502d3103f793e863c44c6bdc7eca93). Closing. In case it is not ok in the next release (2.4.0-alpha2), let us know. Thanks.

Seems that the issue also happens when trying to save .gcode if a project file is located on a network drive or to replace a model with another .stl. Opening recent projects from a cached drive also doesn't work, the slicer just freezes.

[image]
[image]

This happens when the drive is unavailable (offline) and Windows works with cached data (located in `%SystemRoot%\CSC`). Programs should see it as a regular drive, but it seems that this doesn't work with PrusaSlicer.