prusa3d/PrusaSlicer #6876: 2.4.0 Alpha_1 Object movement on build plate

FrenzyIncarnate, 2021-09-02:
### Version
_2.4.0 Alpha_1

### Win 10 OS Build 19041.1165

### 3D printer brand / version + firmware version (if known)
Original Prusa Mini - Latest firmware

### Behavior
* _Describe the problem_
 When there are 2 objects on print bed when I move one of them the other one moves to opposite direction same amount. I ve added the 3mf file.
[duplo_mp_top.zip](https://github.com/prusa3d/PrusaSlicer/files/7098567/duplo_mp_top.zip)

* _Steps needed to reproduce the problem_
Just try to move one of the parts while other one is not selected.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
[duplo_mp_top.zip](https://github.com/prusa3d/PrusaSlicer/files/7098567/duplo_mp_top.zip)

FidelCapo, 2021-09-02:
You have probably enable "Auto-center parts" feature in preferences.

FrenzyIncarnate, 2021-09-02:
Thank you for the reply @FidelCapo I did not remember making that change but probably came as default.

closed 2021-09-02 by FrenzyIncarnate