prusa3d/PrusaSlicer #6877: Improve support interfaces by toggled tilting them 45°

spiky2021, 2021-09-02:
### Version
2.3 with modified support

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Craftbot Flow IDEX  XL

### Behavior
* _Describe the problem_ 
  Im glad to see, that the work I started for base interfaces under soluble interface enteres to PS 2.4 and for purge towers 
  as well. Even rectiliner soluble interfaces are selectable now.

  In the mean time this feature allowed me a lot of nice prints. But therefore I proceeded and found that rectiliner infills 
  achieve better surface qualities, than concentric. On the other hand, there is the draw back, that rectiliner infills won't 
  have a defined contour.  Essential for soluble interface to not detache at the contour.

  I tried to use a multipath to combine rectiliner with a contour at first, but it crashes to often during modulation of 
  bottom contacts at the end. And i don't want to touch it.

  Further I recognized, that an alignment of rectiliner infills with below parse supports tends to generate a tension cracking 
  the interface surface. Therefore I started to use rectiliner tilted by 45° to the parse support, toggling its direction by 90° 
  each layer.

  After manyt big prints I can say, I never had as nice surface qualities with soluble support bevor. 
  My fear, that unsupported interfaces lines due to the tilt bend down without a sheat, didn't show up for at least a spacing 
  of 2.5 mm. It workes with linear and grid type support. Especially for grid supports it avoids the problem of last grid lines 
  being tilted by 90°, that leads to many rectiliner lines fully unsupported. Even the problem of messy contours is 
  deminished, which is good for soluble interfaces.

  Therefore I suggest to use rectiliner tilted by 45°and toggled by 90° each layer, as soon as rectiliner is used for interfaces.

* _Steps needed to reproduce the problem_
   It is easily implemented. Exchange the lines: 
   "angel = support_angle" by "angle = support_angle + 45° + 90° * layer_number%2" 
   for rectiliner interfaces in the "Filler" objects. 

* _Expected Results_
   Object surface quality improves. 
   Better contours, then with simple rectiliner.
   No problems with rectiliner interface above grid supports.
   No tension induced cracking of interfaces due to direct alignment with underlying support.
  
* _Actual Results_
   Less good.

_Is this a new feature request?_ Yes

bubnikv, 2021-12-08:
> Unfortunately I saw that the soluble interfaces disappeared bu using snug supports.

Is it fixed in beta3?

spiky2021, 2021-12-09:
Sorry, I didn't replay. I tested it with beta2 or 3 (don't which anymore) and yes it seems to work. Thanks. What about using it for normal support, as well. Sometimes it is better to have some extended support area below, just because the edges won't be as smooth as they need to be for not deforming the  supported surface.

closed 2026-09-29 by prusaslicer-bot[bot]