prusa3d/PrusaSlicer #6878: Don't even get the splash screen when trying to start PS2.4.0a1 on Win10x64 20H2

Sembazuru, 2021-09-02:
### Version
PrusaSlicer-2.4.0-alpha1+win64-202109011441

### Operating system type + version
Win 10 x64 release 20H2

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
I unzipped PS 2.4.0a1 to my desktop and tried running it from there after making sure there weren't any leftover prusaslicer alpha settings folders in %APPDATA%. No response when trying to start any of the 3 executables, not even the splash screen. I tried starting `prusa-slicer.exe` in the CMD shell to see if any errors were reported there and almost immediately got the command prompt back. I checked my overzealous virus checker (Symantec Endpoint Protection) and didn't see any files quarantined (as often happens with random files in FreeCAD), nor anything relevant in any of the log files.

Is there anything I can do to get a crash report to help track this issue down?

#### Project File (.3MF) where problem occurs
N/A

Sembazuru, 2021-09-02:
I _was_ able to launch 2.4.0a1 at home on my personal laptop... Just not my work laptop.

lukasmatena, 2021-09-02:
> Is there anything I can do to get a crash report to help track this issue down?

You can run with `--loglevel=5` command line argument to enable extensive logging to see if it says something.

Sembazuru, 2021-09-02:
Still no joy, the command prompt comes back immediately:

```
C:\Users\CHELLIOT\OneDrive - Agilent Technologies\Desktop\PrusaSlicer-2.4.0-alpha>prusa-slicer.exe --loglevel=5

C:\Users\CHELLIOT\OneDrive - Agilent Technologies\Desktop\PrusaSlicer-2.4.0-alpha>
```
Just out of curiosity I copied the folder to my C:\ root and tried again (wondering if the path length or spaces might be causing an issue) and got the same result:

```
C:\PrusaSlicer-2.4.0-alpha>prusa-slicer.exe --loglevel=5

C:\PrusaSlicer-2.4.0-alpha>

```
 Though this time I did get a quarantine notification from Symantec for the prusa-slicer.exe
[image]

bbinnard, 2021-09-03:
2.4 loads and runs OK on my Win10/64 Home system. Windows Defender does, however, say  it prevented the program from altering memory.

Sembazuru, 2021-09-04:
Ok, It seems part of my symptoms were red herrings. I started PS2.4.0a1 on my home computer where it does work from the CMD shell, and the prompt returns like on my work computer. I'm running on my home computer with `--loglevel=5` and don't know where to find the log. The log doesn't seem to be going to STDIO. Where is the log file being saved on Win10?

Oddly, on my work computer, even though Symantec is reporting a quarantine, the executable file isn't actually being removed from the source folder. But that might be an issue of Symantec and way out of scope for PrusaSlicer... ;-)

bubnikv, 2021-09-06:
@Sembazuru Frankly I am not sure why you guys are using the antiviruses. They do more harm than good and according to @jschuh, former Google Chrome / Chrome OS security expert who is active on PrusaSlicer github, these antivirus tools often open up questionable network channels for their updates, that may introduce significant security risks.

bubnikv, 2021-09-06:
: I'm running on my home computer with --loglevel=5 and don't know where to find the log. The log doesn't seem to be going to STDIO. 

It writes to stdio, but only if you run prusaslicer-console.exe

bubnikv, 2021-09-06:
: 2.4 loads and runs OK on my Win10/64 Home system. Windows Defender does, however, say it prevented the program from altering memory.

I shall check with our content team whether we could manage to sign our Windows development builds.

bbinnard, 2021-09-06:
The latest up to Windows Defender has improved feedback on threats it discovers. Here is one of several identical ones I've got
using 2.4:

[image]

To stop these I just clicked on the Actions button and selected "Allow on device."

jschuh, 2021-09-06:
@bubnikv, I suspect this is happening because of [the first line](https://github.com/prusa3d/PrusaSlicer/blob/e630332e245b4773beea7c77b54c1062c7a3b3b9/src/libslic3r/BlacklistedLibraryCheck.cpp#L37) in this snippet from `BlacklistedLibraryCheck::perform_check`:

```
    HANDLE  hProcess = OpenProcess(PROCESS_QUERY_INFORMATION | PROCESS_VM_READ, FALSE, GetCurrentProcessId());
    if (NULL == hProcess)
        return false;
```

Calls to `OpenProcess` with `PROCESS_VM_READ` are unusual in general, and particularly from software that isn't system installed, so it's the kind of thing that AV software triggers on. (Of course, the dubiousness of AV relying on such signatures is an entirely different discussion.)

Thing is, the code as written will work, but it's not the right way to do it. I assume someone copied code from a process enumeration example and tacked in the `GetCurrentProcessId()` call, but the pseudo-handle returned by `GetCurrentProcess` (literally just the magic number -1) has `PROCESS_ALL_ACCESS` and gets handled specially. So, you'd avoid all the overhead of invoking the object manager and creating a new process handle context, which also has the nice effect of not triggering the AV signatures.

Shorter version, just replace the above three lines with this:

```
    HANDLE  hProcess = GetCurrentProcessId();
```

I should have time to test it later today and post a short PR. If I can, I'll also see if there's anything else in there that's likely to trigger "security" software. Of course, the irony is that at some point you're likely to use this same code to fix breakage from hooks injected by AV and other "security" software, but that is what it is. 🤷🏻‍♂

bubnikv, 2021-09-07:
Merged the fix by @jschuh, thanks for your help and expertise.

Sembazuru, 2021-09-07:
@bubnikv Re: using antivirus - I have no choice on my work laptop, using Symantec antivirus is required by my corporate IT department.

Sorry for not getting back earlier. I (and most of the rest of the US) was off on Monday for a federal holiday so I didn't have access to my work computer. I tried running `prusaslicer-console.exe` with `--loglevel=5` and it worked. I got my config bundle from PS2.3.3 loaded, shut down, and started `prusa-slicer.exe` using the icon shortcut on my desktop (how I first found it failing) and it started without issue.

I did not update my install of PS, nor do I recall any system updates for my OS. But, now it works. Not sure why. I don't know if there is anything interesting in the log, but I'll attach it here.
[prusa-slicer-console.exe_loglevel=5.txt](https://github.com/prusa3d/PrusaSlicer/files/7122889/prusa-slicer-console.exe_loglevel.5.txt)

Otherwise, my issue seems to have fixed itself? If no one else needs this issue open, it may be closed.

jschuh, 2021-09-07:
@Sembazuru, the Symantec message was a reputation trigger. Either an admin at your company approved the software, or (more likely) the binary was automatically uploaded to Symantec after the "detection" event, scanned, determined to be clean, and signatures were silently pushed out to clients reflecting that status. Thus, it's no longer blocked.

The change in #6914 should resolve the MS Defender warning that @bbinnard saw, and ideally will be one less strike against the reputation score for products like Symantec, when they don't already have a signature for the binary (because it's too new). The goal there being that future updates don't get blocked until the AV maker's automatic scanning and signature generation process completes its cycle.

@bubnikv, I forgot to mention one more trick to avoid false positives on AV reputation triggers. Every AV engine pulls from the [VirusTotal database](https://support.virustotal.com/hc/en-us/articles/115002100149-API), which has an API for public submissions. So, you can submit your binaries to VirusTotal in advance (ideally as an automated step in your release scripts) to ensure that it makes it into the AV reputation databases as quickly as possible. You can also check the submission to see if you're triggering false positives in any AV engines.

bubnikv, 2021-09-08:
@jschuh Thanks for the hints regarding VirusTotal database. Frankly I don't understand how the consumers of the VirusTotal could trust the entry pushed by an anonymous publisher. However I believe the reputation should go higher if we just sign our Windows binary builds and we shall do that for the next development cycle (PrusaSlicer 2.5)

bubnikv, 2021-12-08:
The original issue was likely an antivirus issue, which as improved by removing the offending Win32 API calls as suggested by @jschuh. We will either try to publish our development builds at VirusTotal, or we will sign our Windows builds.

closed 2021-12-08 by bubnikv

jschuh, 2021-12-08:
A bit of context if it helps: Most of the reputation based systems weight the history of the authenticode signature pretty heavily, so if it were me I'd definitely get all the Windows builds signed first to have the biggest impact.

Submitting VirusTotal is extra assurance (and coverage for the minority of systems that weight authenticode history less). But maybe even more importantly, building VT submission and tracking into your release process gives you an automated way of detecting when you're getting erroneously flagged.