prusa3d/PrusaSlicer #6881: Improve the Modifier settings and behavior on the model inside

ncsandor, 2021-09-03:
### Version
2.4.0-alpha1

### Operating system type + version
Windows 10 64bit

### Behavior
If I set the fuzzy skin modifier to none on the modifier body, it prints a smooth permiter on the outside, but unfortunately it also makes permiters  on the inside of the modifier outer surface.

It may be possible to a modifier or function in such a way that the modifications in question only modify the outer shell of the main object and do not place walls inside.

[image]

bubnikv, 2021-09-03:
That is a natural request. BTW PrusaSlicer implementation currently behaves the same way as in Cura, including this particular modifier behavior.

foreachthing, 2021-09-03:
Just because Cura does it doesn't mean PS has to, too. This is where PS would take the first step. I find this behavior silly myself.

janecker, 2021-09-03:
+1

I stumbled uppon this issue as well and think it is not what I would expect. With the current behaviour it is impossible for me to select the specific peremiters I want to define as "outside" that should have the fuzzy skin.

Sembazuru, 2021-09-04:
I get a similar behavior if I try to use a modifier to selectively turn on fuzzy skin using a modifier. Here is the modifier:
[image]
Here are the settings:
[image]
And here is the result:
[image]

I get the exact same result if I have Fuzzy Skin set to `External Perimeters` or `All Perimeters`. (Well, exact within the randomness of the application of the fuzzy perimeter.) I haven't printed this yet, but I would expect that internal fuzzy skin area to be weak, possibly letting the fuzzy part break away easily.

Like the OP, I'm on Win10 64bit (not that I would expect the host OS to affect the slicing algorithm...). This is running PrusaSlicer-2.4.0-alpha1+win64-202109011441

bubnikv, 2021-09-06:
We understand the request perfectly. It is just about the effort and priorities. If you read the responses after each PrusaSlicer announcement, "Still no tree support?" is at the top. We will see whether we can implement this request in PrusaSlicer 2.4 final.

Sembazuru, 2021-09-07:
I understand priorities. One of the other projects that I use is KiCAD, and they are even more restricted on developer resources (all volunteer). Please keep this issue open as a reminder and somewhere to attach results to once you get enough round tuits to be able to spare one for this issue. ;-)
https://www.youtube.com/watch?v=L65v17yeSZE

tol2cj, 2021-09-09:
The modifiers in general work this (in my opinion bad) way. When you intersect model with modifier, the modifier outline creates own perimeters inside model. Is it intended behaviour?
This is openRC truggy tire, where I wanted 3 perimeters on inner wall and 2 perimeters on outer wall
[image]
[image]

zeizei79, 2021-09-13:
As fuzzy skin is affecting only external perimeters, could it be implemented as paint-on feature?

Doprintityourself, 2021-09-16:
I understand priorities as well, but just an experience on that, 
Tried other things with overlaying parts or modifier and it resulted in this double walled parts.

The big problem with this , and so it is not only a fuzzy thing problem, it is a mor common problem, the part gets week with this double borders.
I had one or two that broke becaus the borders just went through a body partially.

Just wanted to mention that this is a problem in other cases two
Thanks

jschuh, 2021-12-15:
I just added a potential fix for the fuzzy skin issue to #6751. The change disables the fuzziness on perimeters hidden inside the model. The picture below should give you an idea of what that means: _The fuzzy skinned model has different materials for the left and right side; the far right corner is a modifier mesh that has no fuzzy skin; the light brown extrusions are the hidden perimeters between adjacent print regions._

If you're comfortable building Prusa Slicer you're welcome to try out the patch. However, the Prusa team is very busy trying to get 2.4 out the door and I expect they won't have time to look into this until sometime after launch, so please don't bother them about it.

[image]
[image]

pmottram, 2022-01-09:
If you don't want internal walls then at the moment you could make modifier regions (from CAD) that are just 2 layers deep - then you get two fuzzy perimeters and two model perimiters, but nothing else slicing through your model: 
[image]
Managed to do this for lettering too: https://www.prusaprinters.org/prints/115317-pi-zero-tv-case

I expect that for this to work without interrumpting the perimeters, it would need seperate code to the way existing modifiers work specifically for this case (i.e. rather than creating a new region that is stiched into the model, it instead just changes the gcode generated for perimeter lines within the region). Still, as it stands it's quicker (and with smaller model files) than adding the roughness onto the mesh manually in blender so I love it.

Another benefit though of changing the code would that it would then allow fuzzy skin applied in vase mode using modifiers, instead of getting this error:
[image]

markus-i, 2022-09-21:
Still there in Prusa Slicer 2.5.0. And, from the discussion, I don't see that anyone found a better workaround to this issue...

Just for completeness: in my case, I want to do a speaker box that should have its walls filled with concrete (printing 14 mm thick walls at 5% infill; cheaper than printing thick walls at >30% infill...). And of course
a) the shape tends to be a bit convoluted and
b) the additional "fuzzy skin"/modifier internal walls would cut off wall sections from the concrete pour

Thanks for the discussion and workaround suggestions so far - it means I can stop looking for what I'm doing wrong and continue to work with what I have.

DragonFinlay, 2022-09-30:
I wanted to leave a small heads-up that this is not exclusively happening with fuzzy skin, but with any modifier that causes specific changes to the perimeters. A good example is the number of perimeters, PrusaSlicer is not yet able to automatically merge e.g. 3 into 2 perimeters, so it creates separate bodies for 3 and 2 perimeters respectively. See #8809 and #8121.

So when a new algorithm is implemented for this, both the fuzzy skin and other modifiers should stop creating these inner walls.

bond2k, 2023-01-04:
I hope this gets fixed. I wanted to print an articulated toys with fuzzy skin, but without being able to exclude the articulating parts, it's almost impossible to do.

cl0ud95, 2023-02-12:
Hi, faced this same issue when using Cura, so I downloaded PrusaSlicer 2.5.0 hoping that this would be the solution.. I needed to print a tool that would take high tensional forces with a textured grip at one end created by the fuzzy skin.

This issue would most certainly cause the structural integrity of the part to fail because the interlayer adhesion would be affected by the 'inner' fuzzy skin. I'm going to try out @pmottram 's suggestion first as an alternative

Jonny360, 2023-03-12:
+1 for this issue. I want to print with fuzzy skin but have a logo appear unfuzzy using a loaded modifier. Unfortunately it cuts into my model and adds these internal extra perimeters. It would be great if at least for fuzzy skin this could be addressed better.

cfontes, 2023-03-22:
Would be great to have this prioritized, also it would be useful for when we have texture modifiers because the same will happen

andyylin, 2023-04-20:
Yes, very annoying when I want to add perimeters to a region but it ends up separating out the region completely. 🤣

St0RM53, 2023-05-08:
I have the same issue, even if using a modifier to prevent fuzzy skin on the inside walls of the model (which is a pain to do as complex surfaces can't be offset correctly) it creates fuzzy perimeter inside the wall of the model which causes the print to fail.

Also if you are using supports you need to keep them far away from the walls as fuzzy skin on the inside will cause not only the support be joined to the inside skin but also causes the print to fail.

Here's my example part (note how far i need to keep the support walls (xy separation) to prevent issues because of the fuzzy skin on the inside (and this is only with 0.1/0.1mm settings):

[image]

[image]

Here's with with custom modifier "plug". It's the inside volume but CAD won't offset the surface correctly so i just scaled it to 101% to get it touching the inside wall

[image]

And here is the issue with the fuzzy skin..The modifier works but instead of modifying the existing perimeter it creates a new one, so the outside perimeter continues to apply fuzzy skin where the infill should be! This causes the print to fail.

[image]

PLEASE FIX

Aasikki, 2023-08-14:
I'm just trying to print a case with fuzzy skin on the outside, and I need to exclude it from some areas where screws and gaskets etc go. Looks like it's not really possible due to this quirk, as having internal fuzz is definitely a no no as I need this to be as rugged as possible. Went through the effort of creating modifiers in cad for it and was quite a bummer to find this out.

I really hope this can be fixed!

user45876, 2024-03-14:
Just wanted tio chime in - this is still the same issue in 2.7.2 :)
Having it as a paint on option sounds great.

tol2cj, 2024-03-26:
Hello, I just wanted to try non-fuzzy text modifier on fuzzy skinned object. What I realized is that we only need slicer to set the fuzzy skin thickness to 0 mm on intersecting area.
[image]

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena