prusa3d/PrusaSlicer #6888: Post-processing

KnatteAnka, 2021-09-03:
### Version
Version 2.4.0-alpha1+Win64
### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
N/A (Prusa MK2.5S)

### Behavior
new behavior that Post-processing is done before copying item to location gives me problems.

Problem is that my script for Starting Printer, copying and mounting the networkdrive need to be run after the export is done :( 

Is there any way to run a file when the file is exported?
Or is it possible to add a check box old or new behavior. (Run when copied [Checkbox])

xorza, 2021-09-03:
I'd suggest that we have two scripts

    rename current one to `pre-processing`
    add a post-processing one

bubnikv, 2021-09-06:
: Problem is that my script for Starting Printer, copying and mounting the
networkdrive need to be run after the export is done :(

What exactly does your post processing script do?

KnatteAnka, 2021-09-08:
it is used to run command when a file is processed 
1: Start printer
2: Connect to Wifi SD-card
3: Copy rename and copy G-code to Wifi SD-card

bubnikv, 2021-09-09:
Would you please post your script?

KnatteAnka, 2021-09-10:
Here is the script,
As a short hot fix for the alpha i use Post.bat to start a new window of Move_rename.bat and in that i have a 10s delay (To give PS time to copy the Gcode to the export folder but that is not optimal because is the time always the same independent of file size...

Post.bat

```
E:  
cd E:\Export  
start Move_Rename.bat  
```

Move_Rename.bat

```
@echo off

chcp 850
chcp 65001
Color F0 

E:
cd E:\Export

timeout 10
set Number=1000

Echo ------------------Test connection--------------
ping -n 1 -w 50 "192.168.3.74" | findstr /r /c:"[0-9] *ms"

if %errorlevel% == 0 (
    echo Printer is started.
    goto Begin
) else (
    echo Printer is offline.
    echo Starting 3DPrinter
)
Echo -------------------------------
start http://192.168.3.88/relay/0?turn=on
timeout 30
:Begin
Echo Starting SDCARD network
set Number=99
net use X: \\192.168.3.74\DCIM
:Start
Echo -------------------------------
Echo Starting SDCARD Export
for /f "delims=" %%x in (Select.txt) do set Number=%%x
set /a Number= Number-1

set File=None
for /F "delims=" %%i IN ('dir /B *.gcode') DO set File=%%i
Set NewFile=%Number%_%File%
IF EXIST "%File%" (
	echo Copying %NewFile%
	Echo ┌────────────────
	Echo │ %Number%
	Echo └────────────────
	rem echo Copying %NewFile%
	rem Echo ^<ESC^>[31m Copying %NewFile%	[30m
Color F0 
	move "%File%" "X:/%NewFile%"
	echo %Number% > Select.txt
	goto Start
)
Echo -------------------------------
Echo Done Export.
timeout 60
exit
```

Select.txt
`2453`

sej7278, 2021-10-01:
is this the same as #6042 (which was for 2.3.0?!) which one should we be commenting on?

i have a similar problem - post-processing script makes some tweaks to the gcode dualstrusion/fan pwm/retraction and then runs gpx to convert to x3g format for flashforge creator pro.

the problem is the post-processing works on the /tmp file, but PrusaSlicer seems to send the output to `SLIC3R_PP_OUTPUT_NAME` and gpx is run on the /tmp file (some sort of race condition?)

bubnikv, 2021-10-04:
@sej7278 

I don't understand your workflow and your issue.

> but PrusaSlicer seems to send the output to SLIC3R_PP_OUTPUT_NAME and gpx tries to run on a non-existant file (race condition - the SLIC3R_PP_OUTPUT_NAME doesn't exist until post-processing is finished)

Are you running the gpx tool asynchronously wrt. to PrusaSlicer post processing script?

sej7278, 2021-10-04:
No, gpx needs to run on the final file at the end of post-processing, but that file doesn't exist in its final destination until after pp with this change, so you can't run gpx as part of pp

bubnikv, 2021-10-04:
Why don't you run gpx from pp in the temp file and either rename the output file from gpx to the original file name or set SLIC3R_PP_OUTPUT_NAME accordingly? PrusaSlicer will copy the produced file for you to the target location.

sej7278, 2021-10-04:
As that's a hack. I could equally say why doesn't the guy who made the PR save to ssd then copy to his slow nas

bubnikv, 2021-12-08:
We have documented the current behavior
https://help.prusa3d.com/en/article/post-processing-scripts_283913

I think everybody could adjust her / his script to work with the current framework.

Also please note that the current behavior (modifying the G-code in place by the post processing script) was always specified by the Slic3r manual, thus if somebody used the post processing script to rename the G-code, one did not follow the manual. From the manual:

> Each script will be passed the absolute path of the G-code file that Slic3r generates. The script will need to modify it in-place; alternatively it can output the new G-code to a temporary file and then overwrite the original one with it.

https://manual.slic3r.org/advanced/post-processing