prusa3d/PrusaSlicer #6889: Future improvement request

tastyf1sh, 2021-09-03:
Version: 2.3.3+win64

No issues just a suggestion/request.

I myself have difficulties with my memory and managing different projects at once can lead to issues. Forgetting to scale, change temperatures, choose brim or skirt etc etc. An absolute nightmare after a 16 hour print to realise i didn't scale it properly! Time and money lost. 

Maybe there could be a feature that could be turned on or off (off for not annoy the people with a good memory) which flags up before you export the file?

A sort of warning box, "Have you checked your settings?" Or something similar.

Many thanks

Jules

troypouliot, 2021-09-04:
I have this same issue with remembering to scale all parts within a project. I find myself quite often getting ready to print the gcode and either remembering that I forgot to scale or not remembering if I did scale or not.
I would like to see a [scale] flag or token that could be added to the filename output so that you could easily see if the particular gcode was scaled.

murk-sy, 2021-09-06:
Slicing wrong has happened to me too, especially since 90% of the time I use one single configuration.

Since always just having a pop-up that basically says "please disregard me" is not very reliably useful once you get used to it, **I would propose a "blank" setting. That would mean that every time you open Prusaslicer, no settings, filament or printer are selected** (or perhaps having each of the three togglable, especially since a lot of people only have one printer/material). Then ensuring you select the right settings becomes part of the process.

**Exporting model name, scale, position, rotation, and perhaps file modified date into the .gcode file sounds like a good thing to store for long term**, and would allow recreating model layouts for printing. Putting it in the gcode filename would require some sort of decision to be made if not all models are scaled the same, so I'm not sure if that works out as well.

tastyf1sh, 2021-09-07:
Thanks for the comments, it's nice to see I'm not the only one!

@murk-sy I think that's a great idea too, a little more involved. 

I agree that people can "Disregard" those messages but I see it very similar to a warning you get when you're about to close a program without saving a file you were working on.  

I myself almost never ignore a save before closing message. It's just a gentle reminder and something that I imagine can be implanted to the program quite easily?

closed 2026-09-29 by prusaslicer-bot[bot]