prusa3d/PrusaSlicer #6892: MMU Painter on small parts; mmu_segmented_region_max_width not working

SomeoneElse58, 2021-09-04:
### Version
2.4.0 alpha1 win-64

### Operating system type + version
Win10 21H1

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S

### Behavior
I am painting [the Enterprise](https://www.prusaprinters.org/prints/936-star-trek-uss-enterprise-ncc-1701) which has many small and vertically oriented parts for multi-coloring printing. The context of the issue is colored items protruding from supported areas are not cutting far enough into the model to avoid printing in the middle of thin air. I have tried modifying the mmu_segmented_region_max_width setting but it does nothing. 

* _Expected/Actual Results_
[image]
On the right side is the PS painted part with mmu_segmented_region_max_width set to 100. That is also exactly what it looks like when set to 0.
Expected is on the left side where I've split the part all the way through the wall using Fusion 360.

I assume it has something to do with how small the parts are.

#### Project File (.3MF) where problem occurs
[Slicer2.4.0_ALL.zip](https://github.com/prusa3d/PrusaSlicer/files/7109300/Slicer2.4.0_ALL.zip)

bubnikv, 2021-09-06:
@hejllukas

hejllukas, 2021-09-07:
Thank you for reporting it. I looked at it, and the small part is not attached to the rest of the model because of two issues. The first issue is that between the red piece and the rest of the model is a small part that isn't printed because of some issue with merging painted regions. Without this issue, it should look like this. 
[image]

But the bigger impact has the second issue, which is the postprocessing of the multiple edges Voronoi diagram. When this input:
[image]
produces this:
[image]
instead of this:
[image]
This would make the red piece stick better. I will try to fix it.

That mmu_segmented_region_max_width parameter limits the size of the colored regions in favor of the object's base color. This means that the colored regions will be cropped according to the set distance from the perimeter. So the colored regions can only be made smaller with this parameter. For better illustration, I attach two examples.

| mmu_segmented_region_max_width=0 (off) | mmu_segmented_region_max_width=5mm |
| --- | --- |
| [image] | [image] |

hejllukas, 2022-02-07:
Sorry for long time without any responds, but it should be fixed with a4ecf2f2a6d140a25e9b76e90ae5f4cf8457bd91, and it should be part of the upcoming 2.4.1-beta2.

closed 2022-02-07 by hejllukas