prusa3d/PrusaSlicer #6898: Bug: "Background Processing" / Auto-slicing interrupts and resets move tool

MNeMoNiCuZ, 2021-09-04:
### Version
2.33

### Operating system type + version
Windows 7

### Bug:
There's a bug that happens when you move a mesh as the program is auto-slicing.

- As it's done slicing, your ongoing move is undone.
- This is frustrating when trying to correct an auto-arrange mesh layout.
- It may interrupt other tools too, but this frequently happens to me when moving pieces around the plate for layout purposes when I have auto-slicing enabled.

- If the auto-slicing was quicker to access, it wouldn't be as big of a problem. 

(https://github.com/prusa3d/PrusaSlicer/issues/6893)

- Maybe it makes sense to delay the auto-slicing by a second or two (with an option).
- And also have an option to prevent or pause it if another tool is being used.

bubnikv, 2021-09-06:
You are right, the "slicing finished" or "slicing aborted" message sent from the background processing thread to the UI thread seems to collide with the gizmo operation in progress.

@enricoturri1966 @lukasmatena

lukasmatena, 2021-09-22:
Should be fixed with a6261ff for 2.4.0-alpha2. Please, retest after it is released. Thanks.
Closing.

MNeMoNiCuZ, 2022-01-26:
@lukasmatena 
This is still happening to me in 2.40.
You can see in the gif below how moving around is interrupted and cancelled.

[image]

lukasmatena, 2022-01-26:
@MNeMoNiCuZ Thanks for the info, but I just tried in both 2.3.3 and 2.4.0. I can easily reproduce it in 2.3.3, it does not happen in 2.4.0 at all. Can you reproduce it on another PC, or just the one with Win7? Are you using the build downloaded from us, or did you compile it yourself?

MNeMoNiCuZ, 2022-01-26:
Interesting. I've only tested at my Win7 PC and I'm unable to test on a Win 10 machine unfortunately.
The build is the official 2.40 downloaded one from Prusa's website (not Github).

lukasmatena, 2024-02-19:
Still unable to reproduce (2.7.2-alpha2) and nobody else reports this. Closing.

closed 2024-02-19 by lukasmatena