prusa3d/PrusaSlicer #6899: Feature Request: The Arrange feature often creates unoptimized patterns

MNeMoNiCuZ, 2021-09-04:
### Version
2.33

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Feature Request
There is a problem where if I use the Arrange feature when I have several copies of an object and I try to arrange them using this feature, the output is not in a pattern.

This means I often move some items around afterwards.

Repro steps:

1. Right click on an empty build plate and select: Add Shape > Box
2. Right click on the box and select: Set number of instances > 20
3. Press the Arrange button

My results is a pattern like this:
[] X X X X
X X X X X
X X X X X
[] X X X X
[] [] X X []

What I would expect with the case of having several instances of the same object, is this:
X X X X X
X X X X X
X X X X X

murk-sy, 2021-09-06:
This is most commonly an issue when rotating the starting piece and then only printing instances of that one piece (I do it to reliably force internal and external seam position)

14 instances normal
[image]

14 instances rotated 45°
[image]

Expected examples
[image]
[image]

MNeMoNiCuZ, 2026-09-18:
I wonder if this is resolved in v3.
An agent could test all these repros/tickets easily...