prusa3d/PrusaSlicer #6900: Bug: Slicing Presets name recursion

MNeMoNiCuZ, 2021-09-04:
### Version
2.33

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
I have been unable to manually reproduce it, but I was copying / "Save As" a lot of files to make variations of it, and I started getting recursive names for the preset in my "Print Setting". I was using a saved Preset (from the "Save current Print Settings").
See screenshot below.

[image]

rtyr, 2021-09-05:
It was common behavior in older PrusaSlicer versions (till 2.3.0-beta2), so I guess your 3MF project was originaly saved in older PS version.

From 2.3.0-beta2 changelog: 

> When loading the presets from 3MF, AMF or G-code, PrusaSlicer used to create temporary "external" profiles. These "external" profiles were named with the name of the source file and suffixed with the name of the source profile in braces, for example "my_print.3mf (0.15mm OPTIMAL)". Starting with this release, the profiles loaded from 3MF, AMF or G-code referencing and modifying an existing local profile are no more loaded as "external", but the referenced profile is activated and modified with the loaded preset. If the referenced profile does not exist, but the profile refers to a system profile by the "inherits" field, the system profile is loaded and modified instead. This works for all profiles with the exception of multi-extruder printer with multiple filament profiles modified. In that case the first modified filament profile will be loaded as modified, while the other modified profiles will be loaded as "external".

Also see https://github.com/prusa3d/PrusaSlicer/issues/3238#issuecomment-599516843, this is example how recursive preset names originated. 

 I think we can close this issue.

MNeMoNiCuZ, 2021-09-05:
Agreed. This is likely. The files I had there are from an older version a couple of months ago.
Thanks for confirming @rtyr

closed 2021-09-05 by MNeMoNiCuZ