prusa3d/PrusaSlicer #6901: Feature Request: Have the left-side toolbox selectable without needing to select a mesh first

MNeMoNiCuZ, 2021-09-04:
### Version
2.33

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Feature Request
In Prusa Slicer, you have to click on your model first, then choose the tool you want on the left side.
In many 3D applications, you can choose the tool first, then you choose which model to apply the tool to.
So often I find myself wanting to pick the Rotate tool, and then select which model to rotate, but this workflow doesn't work in Prusa Slicer.

fe60, 2021-09-15:
Don't only allow to select a tool before selecting an object, but also **keep the last tool selection when no object is selected**.

That way I can use the same tool on several objects even when (nervously) clicking on the plate while trying to select the next object or rotate the view before doing so. The click on the plate resets the object selection and with the current behaviour of PrusaSlicer this results in losing the tool selection.

MNeMoNiCuZ, 2021-09-15:
+1 on the comment by @fe60

MNeMoNiCuZ, 2026-09-04:
This topic is still valid, but with top menu instead.
Oh how I hate the new top menu.