prusa3d/PrusaSlicer #6903: Compare presets window titles inaccurate - 2.4.0-alpha1

JohnOCFII, 2021-09-04:
### Version

Version 2.4.0-alpha1+x64
Build:     PrusaSlicer-2.4.0-alpha1+x64-202109011420
### Operating system type + version

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Big Sur Version 11.5.2 (Build 20G95)
Total RAM size [MB]: 17,180MB

### 3D printer brand / version + firmware version (if known)

Prusa MK3S or Railcore (printer not relevant)

### Behavior

With the new ability to compare configurations, the titles for the different configurations show "old value" and "new value" which does not make sense in this comparison context.  Using the same nomenclature as the parent window "left preset value" and "right preset value" would make more sense.  (See the second screen shot to see the inaccurate titles).

_Is this a new feature request?_

No.

#### Project File (.3MF) where problem occurs

Can see this without a model loaded.
[image]
[image]

rtyr, 2021-09-05:
True, we should rename it, thanks.

bubnikv, 2021-09-25:
Fixed in PrusaSlicer 2.4.0-alpha2

closed 2021-09-25 by bubnikv