prusa3d/PrusaSlicer #6904: 2.4.0 alpha1 allows user to export G-code even though the UI as "Generating G-code..." but doesn't export anything unlike 2.3.3

gudnimg, 2021-09-05:
```
PrusaSlicer
Version:   2.4.0-alpha1+win64
Build:     PrusaSlicer-2.4.0-alpha1+win64-202109011441

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 16,349MB
OpenGL installation
GL version:   4.6.14822 Compatibility Profile Context 20.50.26.01 27.20.15026.1004
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon(TM) Graphics
GLSL version: 4.60
```

### Behavior
There is a problem with Alpha 1 where it is possible to generate the G-Code while it is being generated, this issue is also present on 2.3.3. But Alpha 1 the operation doesn't actually export any G-Code while 2.3.3 does, so that leads me to believe this is a bug.

It is hard to read in the buttom left corner but it says: `Generating G-code...` and I am able to export the gcode while it is generating at any time: 

[image]

If I try to export the G-code and save it, nothing happens. But after the G-code finishes generating I get this message:

[image]

This file is never created.

But on version 2.3.3 the file is created after the generation finishes (though the dialog shows incorrect file name):

[image]

#### Project File (.3MF) where problem occurs

.3MF file:
[FULL_sq_sunglasses2.zip](https://github.com/prusa3d/PrusaSlicer/files/7110723/FULL_sq_sunglasses2.zip)

bubnikv, 2021-09-07:
We believe you described a behavior that was always there in PrusaSlicer. 

When you press export, the slicing was not finished yet, thus the time estimate is not there yet. Thus PrusaSlicer produces a file name with the time placeholder. It may be confusing, but we think the behavior is correct.

It is not quite correct though, that the file name with the print_time placeholder is shown in the "Eport finished" notification. We believe though that the correct file with the  print_time placeholder replaced is being exported. We may look into the notification.

We will do some more tests regarding your assumption, that in some scenarios the G-code file is not being exported at all. It seems to work for me. Would you please test that one more time?

gudnimg, 2021-09-07:
> We will do some more tests regarding your assumption, that in some scenarios the G-code file is not being exported at all. It seems to work for me. Would you please test that one more time?

No problem, I will take a another look this evening.

gudnimg, 2021-09-07:
Hmm I could not reproduce this again. Maybe it isn't an issue at all 🤔

closed 2021-09-07 by gudnimg