prusa3d/PrusaSlicer #6906: scar "rear" alignment + paint regression

Moyrn, 2021-09-05:
### Version
2.4.0-alpha1 Win64 

### Operating system type + version
Win10 20H2

### 3D printer brand / version + firmware version (if known)
MK3S+MMU2S

### Behavior
scar alignment incorrect : Using "rear" + painting used to be OK in 2.3.0

now in 2.4.0 alpha1 the scar is not aligned, remains in the "painted" area, but is no more rear aligned
[image]
[image]
[image]

now the same 3mf file opened in 2.3.0 : all is aligned and fine
[image]

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
[BEIS12-B.zip](https://github.com/prusa3d/PrusaSlicer/files/7111595/BEIS12-B.zip)

rtyr, 2021-09-07:
Should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/298825672bf985b7467fe4cb033db1835d9abd88. Closing.

closed 2021-09-07 by rtyr

Moyrn, 2021-09-07:
pretty neat - thank you