prusa3d/PrusaSlicer #6908: Support Brim

Tommy-Mus, 2021-09-05:
### Version
Version 2.4.0-alpha1+win64

### Operating system type + version
Windows 10 Home - 21H1

### 3D printer brand / version + firmware version (if known)
MK3S+ and MINI+

### Behavior
The brim is no longer added to the base of supports, whereas in previous versions (Version 2.3.3+win64) this was added

#### Project File (.3MF) where problem occurs
Any project

rtyr, 2021-09-05:
Yes, it is mentioned in the changelog. 

> Brim is newly calculated around the object, not around the 1st layer of the support and supports are printed over the brim on other layers #713 #1156 #3395 #3396.

bubnikv, 2021-09-06:
Intended behavior, closing.

closed 2021-09-06 by bubnikv