prusa3d/PrusaSlicer #6909: [Bug] "Detect Thin Walls" changes perimeter order

ForbiddenEra, 2021-09-05:
### Version
v2.33

### Operating system type + version
Win 10

### Behavior
When selecting "Detect Thin Walls", it seems to change the order of Perimeters/External Perimeters. I have a file with an overhang (in the middle area) and when selecting this option, the external perimeter is printed before the regular perimeter, which makes overhangs more of a challenge as they only contact the layer below, instead of the perimeter beside it and the layer below.

In this screenshot, we can see that the internal perimeter (yellow) is printed before the external perimeter (orange) on this layer:
[image]

Now, if we select Detect Thin Walls, the order gets reversed, and you can see the orange perimeter in the middle printed before the yellow:
[image]

Even if a perimeter line here is being collapsed into another (regardless of whether it's internal or external) I would expect the order to not be changed, unless of course I selected "External Perimeters First", which is not the case.

Been struggling for a while trying to figure out why I've been getting mixed results on this, I haven't tried a print without that setting yet but just looking at the g-code I strongly believe it will have a better chance of higher quality as the orange external perimeter overhang it will be able to attach to the yellow internal perimeter, instead of just printing half-hanging off the layer below.

ForbiddenEra, 2021-09-05:
Perhaps we could have a "force internal perimeter first" thing? even though disabling Detect Thin Walls helps and there's at least one internal perimeter for the external to attach to, it doesn't necessarily print all internal perimeters before external, as you can see here, while the inner perimeter (yellow) did at least print one line for the external perimeter (my overhang) to attach to, it's still printing internal perimeters after that point - so while it does have one line to attach to, I think the best result would be if the internal perimeter completely finished before the external - basically the opposite of "external perimeters first" ..? 

Honestly having the External Perimeters First thing sort of implies that if it's not checked that the internal perimeters would be printed first but this isn't the case.

[image]

ForbiddenEra, 2021-09-06:
I should also note that aside from that, depending on the layer height / extrusion width, the inner/external order can also change..

such a headache

ForbiddenEra, 2021-09-07:
After more evaluation, I don't think the order change is necessarily just caused by the Detect Thin Wall options so much as whatever changes that option is doing is affecting the order, this is clear as changing the perimeter widths seems to affect order as well.

dartrax, 2022-07-06:
This is still an issue with 2.5.0 alpha 2, but only when Perimeter generator is "Classic" (not Arachne).

Here is an example project:
[DetectThinWalls.zip](https://github.com/prusa3d/PrusaSlicer/files/9054875/DetectThinWalls.zip)

dartrax, 2026-09-18:
This bug is also present in 3.0.0 alpha.