prusa3d/PrusaSlicer #6911: Feature Request:  Smart Fill for Supports

enderjones, 2021-09-06:
### Version
2.4.0 Alpha1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S
Prusa Mini+

### Behavior
Please add smart fill to supports similar to the new multimaterial painting.  If it is all 1 face, paint the whole face for supports.  This will make adding custom supports much easier.

Is this a new feature request?
Yes

strayr, 2021-09-22:
I came here to make the same FR[ after this conversation ](https://www.reddit.com/r/prusa3d/comments/psg8l7/is_there_anyway_to_set_manuel_supports_with/hdvkxgc/?context=3)

hejllukas, 2021-10-07:
Yes, you are right. Smart fill (from the multi-material painting gizmo), in some cases, could make adding custom supports much easier. In commit d2cd14ba7ddf5773d68ee4a8cb0cb028984a323d, we ported smart fill, possibility to enable/disable triangle splitting and the brush for selection one facet, from multi-material painting gizmo to FDM support painting gizmo.
It should be included in the next Alpha release.

closed 2021-10-07 by hejllukas