prusa3d/PrusaSlicer #6912: Feature Request: Arrange Sequentially

MNeMoNiCuZ, 2021-09-06:
### Version
2.33

### Operating system type + version
Windows 7

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Feature Request
I think it would be great to have in the "Arrange Options" (when you right-click the Arrange-button), a checkbox to change the spacing to whatever space is needed to print the objects sequentially. It could also ask if the user wants to enable Sequential Printing at the same time, similar to how it asks you to change settings when you turn on supports or set it to 100% infill.

This would work great with the new and updated visualization of the Sequential Printing.

rtyr, 2021-09-07:
PrusaSlicer already does that automatically (clearance is taken in account). Arrange in combination with sequential print could certainly work better in general, but that is covered in different issues.

[image]

With Complete individual objects enabled.
[image]

MNeMoNiCuZ, 2021-09-07:
Ah, that's nice. I didn't know about this. My bad.

Could it still be worth having a "nudge" to sequential printing there? I didn't know it was a feature from "browsing" the application.

I guess the new feature of "did you know...." will help teach users about the software in a better way.

DragonFinlay, 2022-09-23:
Closing since the original issue is answered and solved

closed 2022-09-23 by DragonFinlay