prusa3d/PrusaSlicer #6915: Cannot import .sl1 files into PrusaSlicer

Aligorith, 2021-09-06:
### Version
Version 2.3.3+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa SL1

### Behavior
Trying to load/import a .sl1 file into Prusa Slicer 2.3.3 fails.

Steps to reproduce:
1) File -> Import -> Import SL1 / SL1S archive
2) Browse for a .sl1 file (see notes below)
3) Select "Import Model and Profile" or just "Import Model Only"
4) Leave "Quality" on "Accurate"
5) Click "Ok"
6) The import fails with the following error: "Invalid SL1 / SL1S file" 
[image]

#### Project File (.3MF) where problem occurs

Tested with the following files:
* .sl1 files from the `"Sample Gcodes / SL1"` folder installed by the PrusaSlicer install wizard
* `original_prusa_sl1_calibration_test_object_v2.sl1`  - [original_prusa_sl1_calibration_test_object_v2.zip](https://github.com/prusa3d/PrusaSlicer/files/7118008/original_prusa_sl1_calibration_test_object_v2.zip)

rtyr, 2021-09-07:
I think the problem is in that particular sl1 file, prusaslicer.ini is not included there.

Does it happen with different sl1 files? If not, I think we can close this issue.

Aligorith, 2021-09-07:
Ah I see.  As noted, the other ones that fail are some of the example ones bundled with PrusaSlicer (i.e. the ones installed to `C:\Users\Public\Documents\Prusa3D\Sample Gcodes`).
 * Those in the `SL1/` folder do not load
 * Those in the `SL1S/` folder *do* load

It seems that with the ones not loading, they all lack the prusaslicer.ini file.

In what version of PrusaSlicer did SL1S support get added?

rtyr, 2021-09-07:
It looks like we have problem with opening SL1 files generated by old PrusaSlicer versions. We will improve backwards compatibility. Thank you for reporting this problem.

commit 62951455 (2021-09-21): Switch to current SLA profile if an imported archive doesn't contain it.

Send a warning notification if this happens. Also,if  there is no current SLA profile initialized, an error message will tell the user to switch to an SLA profile before importing.

fixes #6915
commit 4d2bf8db (2021-09-27): Switch to current SLA profile if an imported archive doesn't contain it.

Send a warning notification if this happens. Also,if  there is no current SLA profile initialized, an error message will tell the user to switch to an SLA profile before importing.

fixes #6915
closed 2021-09-27 by tamasmeszaros