prusa3d/PrusaSlicer #6919: [FEATURE REQUEST] Alternativ Icon for Alpha/Beta/Rc

ncsandor, 2021-09-07:
### Version
2.4.0 Alpha1

I would love to see a different color icon for the alpha, rc, beta. Maybe where the orange is, it could be blue. Working on the taskbar makes it easier to distinguish the software under development from the stabile. I know I could probably  solve this myself, but I would love it if it were an official thing.

[image]
Which is stable, which is beta?

[image]
I mean something like that.

Area5142, 2021-09-07:
I agree, it would make it easier to distinguish between different versions. Blue icon is used by SuperSlicer so that would be confusing.
I added the letter A/B to the PrusaSlicer icon on my Linux system to help identify different versions:
[image]

The blue is SuperSlicer, A marks Alpha version (PrusaSlicer 2.4.0A1) and normal orange icon is the regular released PrusaSlicer.

fe60, 2021-09-07:
Just another proposal based on https://en.wikipedia.org/wiki/File:Greek_letter_alpha.svg
[image]

ncsandor, 2021-09-08:
I made few more variations.
"Summoning bubnikv" You just have to choose. =D

[image]
[image]
[image]
[image]

ncsandor, 2021-09-08:
> I agree, it would make it easier to distinguish between different versions. Blue icon is used by SuperSlicer so that would be confusing.
> I added the letter A/B to the PrusaSlicer icon on my Linux [...]

I haven’t even heard of this super slicer, it made me curious.

Area5142, 2021-09-08:
@ncsandor 
SuperSlicer is based on PrusaSlicer by Prusa Research and can be found [here on GitHub](https://github.com/supermerill/SuperSlicer).
It has everything that PrusaSlicer has and some more, mostly advanced and expert settings.

bubnikv, 2021-09-08:
@Area5142 
> It has everything that PrusaSlicer has

Are you sure? How about PrusaSlicer 2.4.0-alpha1? How can you be sure that SuperSlicer contains all PrusaSlicer's features and in the same quality?

fe60, 2021-09-08:
@ncsandor

> "Summoning bubnikv" You just have to choose. =D

Creating/providing the icon(s) is only one part of the overall process. Making icons depend on the version of the release requires additional effort.
[image] [image] [image]

Area5142, 2021-09-08:
@bubnikv sorry for using the word _"everything"_... I should have used the word _"similar"_ instead. It is difficult to make a very short description for a very complex program.

I am a fan of PrusaSlicer and use it all the time. Including alphas, betas, RC and self compiled versions.

The description for SuperSlicer says that it is: _SuperSlicer is based on PrusaSlicer by Prusa Research_. Some of the features is very handy, like the build in filament calibration prints and guides.

The main topic here is the proposal of different icons for alpha/beta/RC of PrusaSlicer to avoid mixing them up on the Desktop and in the Task bar. I have the problem and edited different icons to solve the problem. Blue was already taken by SuperSlicer.

bbinnard, 2021-09-09:
Here are PrusaSlicer icons for Alpha & Peta versions. 
[image]
[image]

 There are programs that replace icons into existing EXE files. The one I used is icofx which is free for the first 30 days. There are probably others that are simpler and easier to use.

[PSIcons.zip](https://github.com/prusa3d/PrusaSlicer/files/7133589/PSIcons.zip)

mikolaszuza, 2021-09-12:
This does not seem like a bad idea.
@bubnikv would be implementing this complicated? If it's reasonably easy, I can prepare the icon files for you. (something along the lines of what was already shown here)

closed 2026-09-29 by prusaslicer-bot[bot]