prusa3d/PrusaSlicer #6920: Can't set fanspeeds to temperature tower test

Rudis1261, 2021-09-07:
### Version
Version 2.3.2+win64

### Operating system type + version
Window 10 64bit

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro running Klipper

### Behavior
I am trying to set fanspeeds at different layer heights to tune my ABS profile. But it's incredible frustrating. The only fanspeeds that apply, is what the material set. And it sets it on every layer. So I can't even work around it.

I have tried setting it via custom GCODE, I have tried setting it via machine GCODE and scripting. It's impossible to  make any progress tuning my machines profile. 

#### Project File (.3MF) where problem occurs
[temptower.zip](https://github.com/prusa3d/PrusaSlicer/files/7123440/temptower.zip)

rtyr, 2021-09-07:
It should be easy to do. Just set all fan speeds to 0% in filament profile (including bridging speed), then add custom g-codes (M106 S0-255) using vertical slider. Can we close the issue?

[image]

[image]

Rudis1261, 2021-09-07:
Thank you!! I must have tried everything except that 😄😄

You may close the issue!

Kind regards,
Rudi

closed 2021-09-07 by rtyr

Rudis1261, 2021-09-07:
[image]
😌 🙏  Thank you