prusa3d/PrusaSlicer #6922: Can't get PrusaSlicer to print more than 4 layers

Radtaz, 2021-09-08:
PrusaSlicer Version 2.3.3
I just installed it and set it up by following the guy on The First Slice. It's working correctly with Cura but I want to use Prusa Slicer because I like your attributes such at different infills
and 

I'm using Windows 10 64 bit 

my printer is a Creality 3D Max running Marlin 4.2.2

### Behavior
it prints about 4 layers and stops extruding filament. it doesn't realize it has stopped extruding filament and goes thru every layer and completes. 

I just started using Prusa

rtyr, 2021-09-09:
Please provide your project file (File -> Save project as...).

Radtaz, 2021-09-09:
I don't know what file you are talking about. I can't go back and find a print that I canceled when it failed.

Radtaz, 2021-09-09:
I guess I will have to recreate the issue tomorrow using PrusaSlicer after I'm done with the 20 hour print I'm doing with Cura right now. Then I will attach file. Hope you can help because I really like the feel of PrusaSlicer

Radtaz, 2021-09-09:
> Please provide your project file (File -> Save project as...).

> Please provide your project file (File -> Save project as...).

I guess I will have to recreate the issue tomorrow using PrusaSlicer after I'm done with the 20 hour print I'm doing with Cura right now. Then I will attach file. Hope you can help because I really like the feel of PrusaSlicer

Radtaz, 2021-09-09:
> I guess I will have to recreate the issue tomorrow using PrusaSlicer after I'm done with the 20 hour print I'm doing with Cura right now. Then I will attach file. Hope you can help because I really [...]
I think I got a project file attached.

[20mm_calibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/7140278/20mm_calibration_cube.zip)

rtyr, 2021-09-10:
I am afraid the issue is on the printer side, the g-code generated from your project is ok (I even printed it on different printer). 

Is it stock printer with stock firmware? Maybe your print fan cools the nozzle bellow minimal temperature threshold (cold extrusion prevention).

[image]

Radtaz, 2021-09-10:
I printed a calibration cube myself w/Cura Slicer and it worked

rtyr, 2021-09-10:
> I printed a calibration cube myself w/Cura Slicer and it worked.
Probably some difference which doesn't trigger issue with your printer. For example, Cura increases print fan speed gradually by default (similar to our "Full fan speed at layer" setting).

rtyr, 2021-09-16:
No response. G-code is ok. Closing.

closed 2021-09-16 by rtyr

Radtaz, 2021-09-19:
Sorry, didn't get back to reply. Had to start working back in office last week and didn't have a lot of time to play around with this. Yesterday I compiled latest Marlin firmware for my printer adding CR Touch and now I printed a Benchy and PrusaSlicer worked great. So I don't know if the new firmware solved the problem or it was something else. But I don't really care as long as it's working and continues to work. Thanks for trying to help