prusa3d/PrusaSlicer #6924: Modifier object not overriding 1st layer speed

troypouliot, 2021-09-09:
### Version
Version 2.4.0-alpha1+win64 (same behavior in version Version 2.3.3+win64)

### Operating system type + version
Windows 10 64 bit

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
In order to promote better bed adhesion in certain difficult geometry scenarios, I would think I could apply a modifier object to the select areas that is the same thickness as the first layer and change the speed lower. 
It appears that the "First layer speed" setting is overriding the modifier object. I think it should be the other way around where a modifier object should override any other similar settings.

A bonus fix would be having modifiers like this that don't change the slicing geometry not break the model into separate parts with individual and separate perimeters.

Modifier object placed with the intent to reduce the print speed in this area on just the 1st layer;
[image]

Main speed setting;
[image]

Speed settings for the modifier object;
[image]

Actual print speed when sliced (1st layer). The main "First layer speed" is set to 20mm/sec. The modifier object is set to 5mm/sec;
[image]

_Is this a new feature request?_
I guess so?

#### Project File (.3MF) where problem occurs
[Modifier object speed issue.zip](https://github.com/prusa3d/PrusaSlicer/files/7137131/Modifier.object.speed.issue.zip)

rtyr, 2021-09-09:
First layer speed setting overrides all print movement speeds in general (even without modifiers).

troypouliot, 2021-09-09:
Yes, I understand that. I think if the user places a modifier, that modifier should take precedence over all other settings though.
Would there be a reason not to?

rtyr, 2021-09-09:
You would need modifier with `first layer speed` setting override, such setting is currently not available for modifiers. With current implementation, using such modifier would break the perimeters (also visible on your picture).

troypouliot, 2021-09-09:
Yes, having  `first layer speed` available as a modifier setting would suffice. This is the main thing I'm after. The other item about not breaking the perimeter when not needed is just a "nice to have" request. Thanks

troypouliot, 2021-09-10:
FWIW, SuperSlicer seems to support this exact behavior even though `first layer speed` is not a modifier option. If I add `external perimeters speed` and `default speed` to the modifier object it seems to override the main `first layer speed`

[image]

closed 2026-09-29 by prusaslicer-bot[bot]