prusa3d/PrusaSlicer #6926: [bug] Checkbox columns in modal windows are stretched  (macOS)

Hologos, 2021-09-10:
### Version

**Version:** 2.3.3+x64
**Build:** PrusaSlicer-2.3.3+x64-202107161034

### Operating system type + version

**Operating System:** Macintosh
**System Architecture:** 64 bit
**System Version:** macOS Catalina Version 10.15.7 (Build 19H15)

### 3D printer brand / version + firmware version (if known)

Creality Ender 3 V2

### Behavior
*  _Describe the problem:_ In all modal windows the very first column (with checkboxes) is stretched and makes the second column (with text) clipped. Upon resizing, it fixes itself.
* _Steps needed to reproduce the problem:_
  * Click on model, add modifier block.
  * Click on `Add settings > Advanced`.
* _Expected Results:_ Readable description of checkboxes.
* _Actual Results:_ Description is clipped.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred:_
[image]

_Is this a new feature request?_ No

YuSanka, 2021-09-13:
Thanks for your report. This bug is fixed and will be a part of next Alfa

closed 2021-09-13 by YuSanka

Hologos, 2021-09-13:
Thank you for the fix. Love the slicer!

commit 5a957949 (2021-09-13): OSX specific: Improvements for wxMultiChoiceDialog: Height of a ChoiceListBox will respect to items count
This improvement fixed #6926 - Checkbox columns in modal windows are stretched (macOS)