prusa3d/PrusaSlicer #6927: [FEATURE REQUEST] Please add a way to change sequence of objects without having to move them around

Ali229, 2021-09-10:
### Behavior
If I have multiple objects and I want to change their positions then I have to write down their positions (X,Y,Z) and then swap them one by one.

### Requested Behavior
1. Press E
2. Click sequence number
3. Define sequence number

rtyr, 2021-09-10:
You don't have to change the position at all, use object list on the right to change the sequence.

[image]

Ali229, 2021-09-10:
I was moving one object at a time, but putting them one over the other in the object list does the job, thank you!

closed 2021-09-10 by Ali229

ssill2, 2021-09-10:
does this require you to complete each object before moving on to the next.  I've tried to set up the same test case but having six cylinders.  The behavior that I would expect is that for each layer, the object would be visited in the order they are on the right.  Say I define the order as 1, 3, 6, 5, 2, 4 and I slice the .3mf.   The first object it starts to print is 5 in my case.