prusa3d/PrusaSlicer #6928: [Feature Request] Minimum required area for ironing, and / or a paint on ironing blocker.

beggernator, 2021-09-10:
Version:
2.3.3 

Operating system type + version:
Windows 10

3D printer brand / version + firmware version (if known):
Creality ender 5 plus, custom marlin.

Behaviour:
I was wondering if a minimum required area (or paint on ironing blocking) could be added to the ironing feature.
I feel this would be useful as it would allow for only large sections of the model to be ironed without wasting time on smaller areas of the same part that would see no benefit (or even a detriment) from ironing.

Attached is an image hopefully demonstrating the issue this would be solving. (red are for areas that see no benefit from ironing, and thus shouldn't be, and blue is for ones that should be ironed.)
[image]

I presume it would work much like the minimum required area for infill.

Thanks for the time / consideration.
Tom

Is this a new feature request?
Yes

edwardhorsford, 2021-09-13:
I'd also like to see this. I tried ironing this weekend on a small part - whilst a largish area had a great appearance, the ironing on smaller details made those details much worse - to the point that I had to disable it entirely.

My ideal would be a minimum area needed for ironing, and a paint-on brush.

dartrax, 2021-09-21:
I think using a height range modifier to enable ironing only on two or three layers would be easier than paint-on-brush in this case. Also, it would survive reload-from-disk.
However, a minimum area for ironing could be handy. Even better would be a minimum diameter of the largest enclosed circle of an area, so that large but thin areas like rings are not taken into account.

bonjipoo, 2021-11-28:
I would also like to see a minimum area for ironing.  The issue being its easier to enable 'all top surfaces' sometimes for a part but at the moment it tries to iron every tiny top surface on a gradient/chamfer face for example or the bottom face of a horizontal hole.  The only way around this is to set up a whole bunch of height range modifiers which omit those layers.

sglux, 2022-05-23:
For me a modifier AND minimum area seems to make sense, but for thin rings a minimum "stroke length" may also make sense.

Never used the "paint-on" modifiers yet - do they flood-fill the selected horizontal plane?

dartrax, 2022-05-23:
Yes, when you select Tool type "Smart fill". You can try that with the Paint-on supports.

Instead of a minimum "stroke length" I would suggest a minimum diameter of the largest enclosed circle of an area.

dartrax, 2026-09-18:
Still a valid FR