prusa3d/PrusaSlicer #6929: [Feature] Allow temperature change with color change

jnweiger, 2021-09-11:
### Version
2.3.0

### Operating system type + version
Linux Mint

### 3D printer brand / version + firmware version (if known)
I3 MKS3

### Behavior
I use color change to print text on white parts. While the filament change itself works great, I wish for
* A custom string that is shown on the display when I am prompted to insert new filament. I'd use it to say e.g. "load green ABS"
* Nozzle temperature setting (or full Filament change). I often mix materials whatever spool I have for the desired color is used.

[image]

As a workaround, the "Height range Modifier" could also be used to tweak parameters for the color change. That'd be fine with me. 
But currently nozzle temperature is not available (speed is, which I often use!)

jnweiger, 2021-09-11:
This aims in the general direction of https://github.com/prusa3d/PrusaSlicer/issues/2002

rtyr, 2021-09-11:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2002, https://github.com/prusa3d/PrusaSlicer/issues/3439, https://github.com/prusa3d/PrusaSlicer/issues/2851. Closing.

closed 2021-09-11 by rtyr