prusa3d/PrusaSlicer #6930: 【BUG】Estimated filament usage differs between sliced info and the feature type percentage box.

5jvm0u4, 2021-09-11:
### Version
2.4.0 alpha-1

### Operating system type + version
Windows 10

### Behavior
[image]
[image]

The total filament usage is the same over this two place but the filament usage estimated for wipe tower is different as seen in the screenshot.
And this happens with and without No Sparse Layer(Experimental) turned on (picture 1 and 2).
Bty, I was using the length to do my calculation.

#### Project File (.3MF) where problem occurs
.3MF is not support, so instead I zipped the file.
[Filament usage bug.zip](https://github.com/prusa3d/PrusaSlicer/files/7147875/Filament.usage.bug.zip)

5jvm0u4, 2021-09-11:
I'm so grateful that you guys add this feature in as I mentioned in issue #5417, it's just so handy to have.

SomeoneElse58, 2021-09-29:
I've also noticed that almost all of the Sliced Info box are wrong; seems like the alpha versions are slightly more than doubling everything except for the estimated printing time.

[image]

[MMU.logoTail.zip](https://github.com/prusa3d/PrusaSlicer/files/7249616/MMU.logoTail.zip)

closed 2026-09-29 by prusaslicer-bot[bot]