prusa3d/PrusaSlicer #6931: 【BUG】Paint color leakage

5jvm0u4, 2021-09-11:
### Version
2.4.0 alpha-1

### Operating system type + version
Windows 10

### Behavior
[image]
[image]
[image]

As shown in the screenshot, I was only painting the outer edge of the model, but somehow the color leaks into the middle pole of the model and appears on the surface.
After some research I notice that tinkering with the mmu_segmented_region_max_width can solve this problem, but I still consider this a bug since the leakage is too deep into the object that it even pierce through the surface of the object, which is probably not one's expecting.

#### Project File (.3MF) where problem occurs
.3MF is not support, so instead I zipped the file.
[Color Painting Leakage.zip](https://github.com/prusa3d/PrusaSlicer/files/7147941/Color.Painting.Leakage.zip)

davbow123, 2021-09-18:
I am having this issue as well.

SomeoneElse58, 2021-09-18:
Thirded. Happening here as well.

[image]
[image]

[WSB_alone.zip](https://github.com/prusa3d/PrusaSlicer/files/7189381/WSB_alone.zip)

hejllukas, 2021-11-02:
@SomeoneElse58 Your issue is the same as #7112, which was already fixed with 7ce13b4a26e69c96296b801d363d0da65a623674.

lukasmatena, 2021-11-12:
This should be fixed in PrusaSlicer 2.4.0-beta2 (will be released shortly). Closing.

lukasmatena, 2021-11-12:
Sorry, correction. Issue reported by @SomeoneElse58 is indeed the same as #7112 and is fixed in beta2.
The original issue reported by @5jvm0u4 is different and is not fixed.

closed 2024-02-19 by lukasmatena