prusa3d/PrusaSlicer #6932: Feature Request - Save PID values along with material's profile

Wyzed17, 2021-09-11:
Hi there,

Since my printer is enclosed, I found that there are noticeable differences between doing or not a PID tuning before changing for a material that has way more (or less) temperature. For example, from PLA (215) to PETG (250). +/- 1º to 2º C of variation makes a wobbly effect on the print.

It's quite annoying to do it the classic way. Maybe you can add the PID settings as a new parameter for a material. Once you do a fine tuning for a given temperature in a specific condition, I should be able to retrieve it from the printer (Pronterface or OctoPrint?) and then add it to the material profile. Thus is not really neccesary to do a PID tuning before each material change, only pass the proper data in the "start" gcode.

Thanks in advance!

rtyr, 2021-09-11:
It is already possible.

[image]

rtyr, 2021-09-13:
Answered. Already implemented. Closing.

closed 2021-09-13 by rtyr

Wyzed17, 2021-09-13:
> It is already possible.
> 
> [image]

Oh wao, you're right. I totally forget that posibility. I usually use custom G-Codes for the machine itself but not for the materials. Thanks!

Answered at all.