prusa3d/PrusaSlicer #6933: Bug with 2.4.0-alpha1: Can't convert polyline with more than two points to a line

SamLefebvre, 2021-09-11:
### Version
**Version:** 2.4.0-alpha1+win64
**Build:** PrusaSlicer-2.4.0-alpha1+win64-202109011441

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042), 64-bit edition

### Behavior

* _Describe the problem_

When slicing this object on thingiverse [stereographic_projection.stl](https://www.thingiverse.com/thing:202774/files), I have this error: _Can't convert polyline with more than two points to a line_

* _Steps needed to reproduce the problem_

Open the file with PrusaSlicer 2.4.0-alpha1, set the layer height at 0.20mm, Printer : Original i3 Mk3S & MK3s+, slice the file.

The bug only happen when the layer height is 0.20mm,
Work if I set the Printer to Creality Ender-3 V2.
Work on any other layer height.
Work on PrusaSlicer-2.3.3

* _Expected Results_
No error.
* _Actual Results_
[image]

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
The .3MF file is in the zip.
[stereographic_projection.zip](https://github.com/prusa3d/PrusaSlicer/files/7148124/stereographic_projection.zip)

rtyr, 2021-09-13:
Reproduced on master. Sliced ok after Netfabb repair.

bubnikv, 2021-09-13:
This seems to be a very rare issue with the Clipper library, which produces
a polyline with three collinear points when clipping a line with a polygon.

bubnikv, 2021-09-20:
Worked around with https://github.com/prusa3d/PrusaSlicer/commit/0a51afa3e64d5367023075f6dcadd81f801bdd25

closed 2021-09-20 by bubnikv